SynthstromAudible/DelugeFirmware #3316: Sticky shift or harware?

philterwa, 2025-01-26:
### Please describe the problem:

My deluge has a mystery. I had some use of Chopin but only dabbled due to time restraints. After watching a video on some of the new features, I wanted to check them out. I was unable to open the master settings (shift+push select) I only got song settings. Then I realized shift didn't work in any circumstances. Furthmore, I've noticed other irregularities. I'll list them here.

Besides shift not working...
(new+kit) pulls up my drum patch folder of which there's a single folder however...
(new+synth) brings me to my synth folder of which there are many and I'm unable to access those.
(custom 1+synth) doesn't access the dx7 folders.

So let's roll back to an earlier firmware. 
Deluge+OLED+V4p1p0+factory+card+contents
still no shift

At this point I might have to find the tiny driver set and look inside. any advice would be appreciated.

### What is the expected behavior?

Unable to interact with the community feature menu due to the combo (shift+press select) won't pull up the master setting menu.

### When did you start noticing the issue?

2 weeks ago but I've done little troubleshooting until this weekend.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

deluge-c1_2_0.bin

### If possible provide the steps to reproduce the issue and upload additional media:

(shift+anything) doesn't produce a response.
(new+kit) pulls up my drum patch folder of which there's a single folder however...
(new+synth) brings me to my synth folder of which there are many and I'm unable to access those.
(custom 1+synth) doesn't access the dx7 folders.

ok-reza, 2025-01-26:
sounds like your shift button isn't registering.

to make sure, boot the deluge (don't have anything open that hasn't been saved yet)
1) press the first 3 gold knob parameter buttons together: LEVEL/PAN + CUTOFF/RES + ATTACK/DECAY
2) this opens up a test mode to test all lights and buttons on the deluge
3) press SHIFT, any pad press will play a sound, if you aren't hearing a sound, or it's inconsistent, then likely your shift pad is worn out and you need to purchase a replacement (replacement pads are found on the synthstrom store)

philterwa, 2025-01-27:
so I did the button combo and it didn't seem to work. I attached a photo of
me using the remote to push the buttons simultaneously.

Thanks for the help

ok-reza, 2025-01-27:
i dont see any image. also, how long have you had your deluge?

philterwa, 2025-01-27:
It's attached to the email.

I got right about the time oled's were released.

I just checked again and the level/pan and comp/reverb buttons don't toggle
on.

I'm beginning to think it's hardware related.

ok-reza, 2025-01-27:
i don't think attaching photos on email works properly. however it does sound like you need to order replacement pads for the deluge. the amount of time you've had it isn't completely abnormal for pads to not trigger. however, usually a button will be able to trigger if you try pressing it harder to see if they can toggle at all, even if they're temperamental or consistent.

philterwa, 2025-01-27:
Yeah I'm responding to Gmail and github. Noob.

I'm thinking some kind of contamination occurred and my plan is to take it
apart and carefully clean the pads and contacts. Hopefully that works.

ok-reza, 2025-01-27:
let us know if that helps. usually this means the contacts/conductivity are worn on the pads side so cleaning may not help, but would love to know if you find some improvements.

opening deluge guide: https://www.youtube.com/watch?v=M5Go-ReaJEw
replacement pads: https://synthstrom.com/product/replacement-pads/

philterwa, 2025-01-27:
I have to find the time to do it until completion. I'm not opening it up
until then.
I'll let you know for sure. Thanks for the assistance.

philterwa, 2025-02-02:
So I had time to open the deluge up yesterday. I used isopropyl alcohol to
clean the gold contacts and back of pads using a lint free swab. I used an
air source to dry off the parts.
Unfortunately, the same problems exist. The volume/pan and
reverb/compressor buttons do not seem to work as well as the shift button.
In a sense the deluge is slightly bricked. It's a bummer because I was
putting together a computer free music kit with the deluge at the center.

I filled out a form with hardware support and I'll wait to hear back from
them.

Thanks again for your help

Craig

closed 2025-02-05 by m-m-adams

ok-reza, 2025-02-10:
you need to purchase new pads from synthstrom and replace them.

philterwa, 2025-02-10:
do you suggest I order the entire assortment of all pads and completely
replace? If you acknowledged that will solve my problem.
You suggested other buttons that also didn't function. I'd like a little
more guidance. I'm willing to repair myself but I've dug into the deluge
once and I'll do it again but to go from we'll replace your stripped torx
screws to buy some new pads is a little vague.

I want to continue to brag about your customer service

I love this machine and I want to be able to use it.

Please be on point.

ok-reza, 2025-02-10:
for clarification, no one on this github represents customer service for synthstrom, we are all volunteers for developing and testing new features for new firmware. my advice and help with you is just from another deluge user. maybe you contacted synthstrom separately and are confusing that conversation with mine since you are adding comments to the github via email.

if you're stripping your screws you should make sure that the torx screw you use for the encoders are with a1.5mm hex/allen key.

if you dont have any issues with the square pads, you don't necessarily need to replace them, just order the round pads and replace all of them. however, since shipping can be expensive from synthstrom, you could buy both rounded and square and replace everything to start fresh. you could even mark the bad pads so that whenever something fails in the future, you can replace some pads with the old pads that were in working condition.

philterwa, 2025-02-10:
I'll stop responding here and find a new route.
Thanks for all the help.