SynthstromAudible/DelugeFirmware #3317: Crazy red lights and frozen after updating nightly build firmware today 27/1/25

Unrest500, 2025-01-27:
### Please describe the problem:

New to this so not sure I'm in the right place but... [image]

### What is the expected behavior?

Never had a freeze issue prior to this nightly build

### When did you start noticing the issue?

After updating firmware: deluge-v1_3_0-nightly+2025_01_27-76d14f4.bin

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

After updating firmware: deluge-v1_3_0-nightly+2025_01_27-76d14f4.bin

m-m-adams, 2025-02-05:
Does the current nightly work?

Unrest500, 2025-02-07:
Hey, I haven't updated since the issue, haven't had any issues other than the red lights. Can update it and let you know.

Unrest500, 2025-02-07:
Just updated with deluge-v1_3_0-nightly+2025_02_07-7297400.bin (nightly-3) and NO flashing lights!

closed 2025-02-07 by m-m-adams

Unrest500, 2025-02-21:
I've gone back to having nightmares with deluge-v1_3_0-nightly+2025_02_21-fffd08f

https://github.com/user-attachments/assets/93beffa0-4a3f-4e14-8a33-0c6b11322d63
[image]