SynthstromAudible/DelugeFirmware #3318: Nightly 1/27 Crash on Song Load

jasonhischier, 2025-01-28:
### Please describe the problem:

A song that loads fine with the 1/13/25 Beta 1.20 firmware does not load with the nightly 1/27 build. See the 1.X.XML file in the attached zip file.

### What is the expected behavior?

I expected the song to load properly and the Deluge to not freeze.

### When did you start noticing the issue?

Immediately after loading nightly build 1/27.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.3.0-NIGHTLY-76D14F4

### If possible provide the steps to reproduce the issue and upload additional media:

1. Hit Load button
2. Navigate/scroll to song 1.X. Here is song 1.X prior to loading; you can see the preview of the song on the grid buttons:

[image]

3. Hit Load button to load song

Result: the Deluge freezes while showing "Loading." and these fun blue and red lights:

[image]

NOTE: I did install a build from Monitus on the Deluge discord feature forum chat so I could test some arpeggiator improvements; however, I believe I created the song prior to his build, but it's possible I made changes and saved it after installing his build. Here's a link to that build: https://discord.com/channels/608916579421257728/1309291575498772531/1330970981795233883. Regardless, I'm able to open other songs fine, just not this 1.X song.

[1.X.XML.zip](https://github.com/user-attachments/files/18578736/1.X.XML.zip)

soymonitus, 2025-03-29:
Could you recreate the song with latest 1.3 beta?

jasonhischier, 2025-03-30:
Hey, thanks for following up; I just now downloaded last night's 1.3 beta release and I see the issue is resolved. Thank you!

dctucker, 2025-05-09:
Gentle bump since I'm also running crashes that looks similar to this when loading existing XML files after flashing anything based on HEAD of `community` branch.

seangoodvibes, 2025-06-22:
Closing as this has been fixed in nightly recently