SynthstromAudible/DelugeFirmware #3321: MIDI Learn issue

BurningTrestle, 2025-01-29:
I have an MPE device on USB , old school MIDI controller on DIN - both controllers can be learned by Deluge but neither actually alter what I've learned the controller to. 
Deluge is powered by PSU.

BurningTrestle, 2025-01-30:
I'm running Chopin 1.2 - 09D26C7 on an OLED Deluge.

m-m-adams, 2025-02-05:
How are you learning them and have you checked your MPE settings?

sapphire-arches, 2025-02-08:
Also, what do you mean by "alter"? No notes are playing, parameters aren't updating, etc?

m-m-adams, 2025-02-10:
Oh if it's sending on MPE channels all CC info is ignored. If you want to use it as an XYZ controller it needs to be on the main channel of the MPE zone (channel 1 or 16) or a non mpe channel

BurningTrestle, 2025-02-11:
thanks m-m-adams!!!

 whilst I don't fully understand what you're saying it does feel like that is move forward!

The XYZ app when I look at it in the Roli software asks which channel I want it to work on - it is set to Channel 1 currently. It doesn't offer me MPE in the pull down so I am assuming the XYZ app is not MPE. 

when you say Main Channel of MPE zone - does this mean i set the both the Lower and Upper to 1.

(Oddly , the Roli Block doesn't always offer me a range up to 16 - I figured this was to do with the way the ROLI apps are written rather than being an issue)

Please let me know if this is taking up dev time - as it appears that it's not a Chopin bug!

m-m-adams, 2025-03-02:
There's some tips here - https://github.com/SynthstromAudible/DelugeFirmware/wiki/MPE-Setup 

If you have it sending on channel 1 and the deluge thinks it's a lower zone that should work, as would having it not be MPE

closed 2025-03-02 by m-m-adams

BurningTrestle, 2025-03-12:
Many thanks m-m-adams!

I thought I had reported back but my comment isn't here!

The long and short of it was that I didn't realise that the Lightblock was send ing MIDI clock continuously (even though the patch didn't need to) I turned off Clocl reception in the Deluge and it worked straightaway!

J