SynthstromAudible/DelugeFirmware #3344: Crash on song load (nightly)

ch000, 2025-02-07:
### Please describe the problem:

I'm getting a crash on the current master. I've got a project with an kit, an audio track and a couple of midi tracks. When i copy a midi track and try to reload the project i'm getting this crash.

[image]

### What is the expected behavior?

I was expecting the deluge to reload the project to the state without the copied midi clip.

### When did you start noticing the issue?

This is happening with the current master (which is at d08a56f9)
But also with this older nightly:
https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/13104017174/artifacts/2524254562

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.3.0-nightly-1f82d07

### If possible provide the steps to reproduce the issue and upload additional media:

Project file:
[B 16.XML.zip](https://github.com/user-attachments/files/18712076/B.16.XML.zip)

Steps to reproduce:
Copy the first midi track and try to reload the project (load->select project)

https://photos.app.goo.gl/XL38fJwfbtNyusiGA

ch000, 2025-02-07:
Might have something to do with Midi Presets. The midi clip in question uses a midi preset. If i copy another midi track, everything works. If i copy the track with the midi preset it crashes.

ok-reza, 2025-02-10:
[image]

i sometimes get crashes on song load too. this error was from reloading the same song with playback off

seangoodvibes, 2025-06-22:
Closing as I think this was fixed recently in nightly. Please re open if it issue persists.

closed 2025-06-22 by seangoodvibes