SynthstromAudible/DelugeFirmware #3360: Recording audio clip from a specific track will sometimes record source inconsistently

ok-reza, 2025-02-10:
### Please describe the problem:

Arming an audio clip to record a specific track will result in a recorded audio sample that doesn't resemble the original source. The timing will be off and the clips length will be dissimilar.

A previous time I encountered this, the resulting audio recording was like an entirely different groove/rhythm while resembling the notes of the original clip

### What is the expected behavior?

Recording a specific clip from an audio clip should perfectly resemble the specific track.

### When did you start noticing the issue?

at least a month ago

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

DD80936

### If possible provide the steps to reproduce the issue and upload additional media:

https://github.com/user-attachments/assets/9ad56ca1-cfbc-42cb-ad05-d4493ae21c0a

the second clip is the original source
the previewed sound in clip 1 is the audio clip recording
the 1st clip playing by itself represents the audio recording AND the clip length causing time stretching.

both the original synth clip and audio clip are 2 bars long, the recording of the synth clip represents a length that is 5/16th the length of the original clip

seangoodvibes, 2025-05-15:
@ok-reza can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

ok-reza, 2025-05-17:
confirmed issue persists in 1.2.1 and latest beta.

basically sounds like the audio clip's recording of a specific track stops anytime there's no sound being generated, so then when it finished recording it's a lot shorter (depending on gaps within the recording) and plays straight instead of having any time of rhythm