SynthstromAudible/DelugeFirmware #3393: Audio export behaves unexpectedly when MIDI is connected.

stephenterhune, 2025-02-18:
### Please describe the problem:

When I use audio export in song view to export a series of clips at a given beats per minute (my song was at 95 BPM), **and I am connected to an external device with a MIDI cable** (MIDI-IN to Deluge), all clips are exported at a very slow BPM. (I didn't measure, but it's something like >8x slower, maybe.)

However, when I disconnect the MIDI cable and try again, all clips export as expected, and with the correct BPM. All I have to do to make this work as expected is to disconnect the cable.

It does not seem to matter what BPM the external device is set to. This occurs while the external device is not playing any sequence (and I didn't test it while the external device was running. Didn't occur to me at the time.)

### What is the expected behavior?

When I use the audio export feature in song view while connected to another MIDI device via MIDI-IN to Deluge, the export should either occur at the BPM set within the Deluge or at the BPM that is being received over MIDI.

### When did you start noticing the issue?

This happened the first time I tried exploring the audio export feature and it took me a while to figure out whether or not it was potentially just user error. It might be worth noting that my troubleshooting was informed by having noticed the MIDI clock sync light was on nest to the Deluge tempo knob. Even if we end up calling this user error, it seems like unexpected behavior.

### Is there a relevant Pull request?

Not that I know of.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

deluge-c1_2_0.bin

m-m-adams, 2025-02-18:
The device must be sending clock for this to reproduce

stephenterhune, 2025-02-19:
That is probably true.

My setup had a syntakt connected to deluge. Syntakt seems to like to send clock while not playing. It didn't seem to matter (for me) what the clock was set to, though. 95bpm clock from syntakt to deluge where I was expecting 95bpm seems like it should have exported at 95bpm. 

Hopefully that's helpful.

m-m-adams, 2025-02-22:
Oh yeah I don't mean it's not a bug, it's a note to save some time for whoever picks it up to fix

PR #3438 (2025-03-02): don't receive midi while exporting offline audio
Fix #3393