SynthstromAudible/DelugeFirmware #3395: Synth sounds completely different during playback, investigating leads to crash

ok-reza, 2025-02-18:
### Please describe the problem:

Bass sound I audition before playback seems to play normally, during playback it's as if the values/parameters get corrupted for that synth. Changing synth mode is unavailable, it behaves like an FM synth despite being subtractive and I noticed that the osc1 level was at a value of 8042 or so instead of something reasonable like 0 or -12.

Entering osc1 transpose via menu leads to crash.

### What is the expected behavior?

Sounds should play as normal, no crashes when poking at parameter shortcuts

### When did you start noticing the issue?

today's nightly 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

b50231f

### If possible provide the steps to reproduce the issue and upload additional media:

https://github.com/user-attachments/assets/953528f4-d278-46c4-b248-a90a800c9cec

m-m-adams, 2025-02-18:
Please share the file, this is not reproducible in general

ok-reza, 2025-02-19:
for 'Commisary New 3' it seems to be related to the sidechain send clip becoming active and affecting the bass where the bass sound gets messed up.

for '8C.Drift 3', the song would simply crash on load or crash during playback and having all clips launched.

[2.18_nightly_problem songs.zip](https://github.com/user-attachments/files/18855894/2.18_nightly_problem.songs.zip)