SynthstromAudible/DelugeFirmware #3400: Documentation Issue - midi_follow_mode.md

stschoen, 2025-02-20:
### Please describe the problem:

The table listing the default CC mappings for MIDI follow mode contained in Appendix A of midi_follow_mode.md appears blank when viewed with the browser in dark mode. Tested with Safari and Firefox 

### What is the expected behavior?

The table should be visible with the browser in either dark mode or light mode

### When did you start noticing the issue?

just now

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

N/A

soymonitus, 2025-02-20:
I cannot reproduce your issue, It looks fine for me

[image]

stschoen, 2025-02-20:
Current version of the file seems OK. The problem occurs with the file linked from the 1.2 documentation.
https://github.com/SynthstromAudible/DelugeFirmware/blob/release/1.2/docs/features/midi_follow_mode.md

closed 2025-10-22 by seangoodvibes