SynthstromAudible/DelugeFirmware #3417: Transpose stereo audio sample changes it to mono

copb, 2025-02-25:
### Please describe the problem:

When I transpose a stereo audio file in either direction, it becomes mono. 

### What is the expected behavior?

Stereo audio remains despite transposition.

### When did you start noticing the issue?

Just noticed this. Not sure if intended behavior or issue. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.3.0-NIGHTLY-5066B27

darwiniandude, 2025-04-25:
I have today observed the same issue. New (approx 3 weeks) Deluge owner also on 1.3. Have you tried different firmware versions? I’m going to try older ones to confirm if not.

sakamoto-network, 2025-04-25:
Changing the sample interpolation to linear fixes it for me.

darwiniandude, 2025-04-26:
Yep, does not occur when sample interpolation set to linear.

stellar-aria, 2025-05-01:
Closing as fixed by #3654

closed 2025-05-01 by stellar-aria