SynthstromAudible/DelugeFirmware #3448: Audio cuts in a kit row if you press the audition pad during playback

baymud, 2025-03-07:
### Please describe the problem:

If you have audio samples loaded in kit rows (with mode set to "cut") and press the corresponding audition pad during playback it will cut the audio of that particular row

### What is the expected behavior?

Audio should not drop out

### When did you start noticing the issue?

Reproducible on 1.2 and 1.3 (but not on 1.1.1)

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

Using either 1.2 or 1.3 firmware do the following: 

- Create a new kit
- Load a long audio file into the kit row
- Set it to cut
- Press play
- Now press the audition pad
- The audio should cut out

seangoodvibes, 2025-05-15:
@baymud can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

I can't reproduce this

baymud, 2025-05-15:
@seangoodvibes I can reproduce it in both 1.2.1 and 1.3, but have found a catch: I'm using fairly long audio samples (20 seconds +) and it only seems to happen if the kit clip length is longer than the sample (ie, if the sample is 1 bar long, but the kit is 2 bars long). If I shorten the clip length so there is no dead space (ie, both at 1 bar long), it works fine