SynthstromAudible/DelugeFirmware #3450: Deluge file browser / multisample loader cannot render ♯ (sharp symbol) and it breaks the ordering of the samples

abstractrealism, 2025-03-07:
### Please describe the problem:

Samples saved for a multisampled instrument that have a ♯ (U+266F) in the filename get rendered as, for example, G_~23 instead of G♯5. The number at the end is also interpreted as the octave so it puts them at the top of the note range instead of where they belong. 

### What is the expected behavior?

Render correctly and in the correct order.

### When did you start noticing the issue?

First time creating a multisampled instrument, so, today. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

1- Add a folder of multisamples that includes some sharps
2- Try to load them
3- They'll be at the end and with the wrong names.

abstractrealism, 2025-03-07:
Different issue but close enough that I won't clutter the issues page unless it's preferred: the multisample range detector ignores samples that are octave 0. Anything below C1 uses the C1 sample, even if C0 is present.

m-m-adams, 2025-03-21:
Sadly not fixable - FAT32 format doesn't support non ascii characters

closed 2025-03-21 by m-m-adams

abstractrealism, 2025-03-21:
Aha, gotcha, guess that makes sense. 
Shall I open a new issue for what I mentioned in the comment about samples from the C0-B0 octave seemingly not working properly? @m-m-adams 
(If someone can repro, can't promise I didn't just do something wrong, don't create multisamples often)