SynthstromAudible/DelugeFirmware #3474: Unmuting a sample changes sound of another sample

FloMingo808, 2025-03-14:
### Please describe the problem:

I've got a reoccuring problem with sample playback in drum kits. It happens in several projects, every now&then (but once it happens, it doesn't go away). In this example, I'm using a kick from a sample pack, and a rim shot from the Deluge's native files.
The kick has a LP filter and sounds as it should, until I unmute the rim shot. Then the kick sound changes - it sounds like its LPF is turned up again, but it doesn't sound exactly like the unfiltered kick either. This even happens when the rim shot's master level is turned all the way down.
I simplified it for this video, which probably explains it best: 
https://youtu.be/rAMihJxmeVY

More info & what I tried so far:
- OLED, Chopin (but I'm pretty sure it happened on earlier community versions, too) 
- it only happens when the kick's LP filter is lowered. When it's turned all the way up (=off), there's no change to the sound when I unmute the rim shot. So it seems to be messing with the filter.
- when I unmute, then mute the rim shot, and audition the kick with its AUD button, the first and sometimes the second hit sound wrong, then it's back to the normal filtered kick sound. This random behaviour makes me think that it's a bug, not a setting I'm missing.
- no modulations on kick, rim, drum kit (neither in automation view nor mod matrix)
- no compression, delay, reverb, fx on anything
- no side chaining on anything
- Release is set to 0 on both samples
- all samples are set to monophonic + "once" play mode
- when I save the kit and load it in a new project, the problem remains
- when I swap out the samples (including to a native kick sample), the problem remains
- all sample hits have the same velocity
- Master VU meter is green, not even yellow (so it's not distorting)
- using Attack on the kick doesn't help, I'd have to set it to a value where it's longer than the kick sound
- I even copied and used a new, faster SD card - same problem

I might be missing something...but it feels more like a bug.

Thanks for your help & amazing work with the community firmware, you're amazing! 

### What is the expected behavior?

Unmuting a sample shouldn't change the sound of another sample while playing or auditioning

### When did you start noticing the issue?

I think I stumbled over this a few times in the last months (maybe half a year?), but always thought it'd be a setting I'm missing. Now that I've simplified and isolated the issue, I think it's a bug.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

I tried reproducing it by creating the same kit with the same samples, but the problem doesn't happen.

m-m-adams, 2025-04-10:
Can you share the kit and samples?

FloMingo808, 2025-04-10:
Sure, thanks! :)

[Rumbla_Problems.zip](https://github.com/user-attachments/files/19682202/Rumbla_Problems.zip)

seangoodvibes, 2025-05-15:
@FloMingo808 can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

Thanks

FloMingo808, 2025-05-20:
@seangoodvibes - unfortunately, yes... I tried both 1.2.1 and today's beta (deluge-v1_3_0-beta+2025_05_20-1beb424), no change :(