SynthstromAudible/DelugeFirmware #3480: Incomplete song listing while loading song

rozengracht, 2025-03-17:
### Please describe the problem:

I want to load a song out of 20 songs (full of small audio clips only) on my card. I press load + song and turn the select knob clockwise. It shows an incomplete list of songs until it starts from the beginning. If I turn the known counter clockwise I see the complete list of songs.

### What is the expected behavior?

Showing complete song list when turning clockwise.

### When did you start noticing the issue?

Since I have a longer list of songs (no idea when)

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

1. I press load + song 
2. Turn the select knob clockwise. 
3. Incomplete list of songs until it starts from the beginning. 
4. If you turn the known counter clockwise you see the complete list of songs.

soymonitus, 2025-04-08:
Is this issue still occurring on latest beta/nightly?

rozengracht, 2025-04-09:
> Is this issue still occurring on latest beta/nightly?

I never tried as I thought no one was working on the bug.