SynthstromAudible/DelugeFirmware #3491: Row in fill mode triggering by itself on the Nightly FW

Bagatour, 2025-03-20:
### Please describe the problem:

I'm wondering if anyone experienced a row on a drum kit in fill mode (a clap sound for example) triggering some of the notes itself on the nightly FW ? when I switch to Chopin its totally fine but if I go back to nightly build its triggering some notes on the fill row again and some other notes on the different synth tracks also. Is it an issue with the unfinished nightly build? Btw I composed the track on Chopin. But also if I compose a brand new track on the latest nightly build, its doing it again. 

### What is the expected behavior?

a kit row in fill mode has to play only when I press the sync-scaling button

### When did you start noticing the issue?

I'm experiencing this issue at least for over 3 months. I'm frequently switching between Chopin and Nightly but the is issue still there

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_03_19-ba32521

davidvandenbor, 2025-04-27:
I'm also using a nightly build (deluge-v1_3_0-nightly+2025_04_02-cc9091b.zip) and I am experiencing the same problem. Setting notes to “FILL” isn't working, or glitchy (sometimes they play, sometimes they don't) The "FILL"s that I had set on notes in the same song on different tracks previous to the current build (Lets say a couple of months ago) DO work! It's only the notes that I try to set to FILL now, those are the ones not responding to it.

davidvandenbor, 2025-04-28:
I noticed something today: it's as if the FILL function is being "culled" in the same way notes are being left out when the Deluge tries to "distribute" the CPU workload across the song. Sometimes the FILLs act like they are **iterations (1 out of 2)** but I clearly set them to FILL. And it decides on its own which notes it's going to FILL and which ones it won't.

My workflow: during song play, I press the notes that I want to FILL, and then use the black select knob to access NOTE EDITOR > NOTE FILL, and then set to FILL.

davidvandenbor, 2025-04-28:
Wait I should explain it more correctly:
- I have two drumkits, Kit A and Kit B
- Both of these kits have FILLs, when I press "sync-scaling" these are triggered
- Kit A has FILLs that I made a couple of months ago
- With a nightly firmware, I try adding FILLs to Kit B
- The notes aren't muted for KIT B when i press sync-scaling (during performance) sometimes they trigger, sometimes they don't
- I now mute KIT A
- NOW all of a sudden the FILLs on kit B start working