SynthstromAudible/DelugeFirmware #3504: E170 error when deleting a clip which is set as the source for an audio clip

neilalbrock, 2025-03-22:
### Please describe the problem:

When deleting a clip which is the "specific track" source of an audio clip, the Deluge crashes with an E170 error.

[image]

### What is the expected behavior?

The Deluge should not crash and possibly reset the source of the audio track to avoid any further issues or unexpected behaviour.

### When did you start noticing the issue?

Just now but I assume this has been present since 1.2.0 was released.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

- Start a new project
- Set the first clip to be a Kit - I just used a factory kit and entered a few triggers
- Add an Audio clip
- Set the source to be a specific track (with FX) and choose the Kit track
- Record a loop onto the audio clip
- Immediately delete the Kit clip

At this point the Deluge should crash with an E170 error. Please note this does not happen if the first clip is a Synth, only if it's set to a Kit.

neilalbrock, 2025-03-22:
I should also mention that while I was testing this and recreating the error, I also experienced issues with the audio recording, similar to those reported in #3360

lburygin, 2025-03-28:
I can add that the same crash with the E170 error happens in this case:
- Make a new song, make a new kit clip
- Make an audio clip with source: specific track, set specific track to created kit track
- Now go to the kit clip
- Change the kit by turning the select knob
- Keep changing the kit by turning the select knob until it crashes with E170 error (usually after two changes)

neilalbrock, 2025-03-28:
Thanks for the feedback @lburygin, can you confirm if the same crash happens when the first clip is a Synth? Or like me, did you only find it happening with Kit clips?

lburygin, 2025-04-04:
Yes, with synths it is fine, no crashes. I have problems only with kits

lburygin, 2025-04-11:
Seems like it is fixed in the latest 1.3 alpha, can't reproduce this bug anymore

closed 2025-04-17 by seangoodvibes