SynthstromAudible/DelugeFirmware #3509: Crash in nightly when loading a song while other is playing

soymonitus, 2025-03-23:
### Please describe the problem:

This is the crashlog I get:

0x200cdf09: NoteRow::addNotesToSquareInfo(long, SquareInfo&, long&, Note**) at :?
0x200ff081: smSysex::handleNextSysEx() at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/storage/smsysex.cpp:820
0x200d0253: Sample::investigateFundamentalPitch(long, long, long*, unsigned long long*, float*, float*, long, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1084
 (inlined by) Sample::determinePitch(bool, float, float, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1607
 (inlined by) Sample::workOutMIDINote(bool, float, float, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1037
0x200d029d: Sample::investigateFundamentalPitch(long, long, long*, unsigned long long*, float*, float*, long, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1162
 (inlined by) Sample::determinePitch(bool, float, float, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1607
 (inlined by) Sample::workOutMIDINote(bool, float, float, bool) at /Users/raul/Development/Projects/Deluge/DelugeFirmware/src/deluge/model/sample/sample.cpp:1037

### What is the expected behavior?

no crash

### When did you start noticing the issue?

Nightly from some days ago

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

soymonitus, 2025-03-26:
Not sure if the same crash or not but I got more crashes specially with these 2 songs I attach here

soymonitus, 2025-03-26:
0x200ceabd: NoteRow::clear(Action, ModelStackWithNoteRow, bool, bool) at ??:?
0x201000a1: ResizeableArray::getElementAddress(long) at /src/deluge/util/container/array/resizeable_array.h:53
 (inlined by) MultiRangeArray::changeType(long) at /src/deluge/storage/multi_range/multi_range_array.cpp:72
 (inlined by) Source::setOscType(OscType) at src/deluge/processing/source.cpp:261
0x200d0e07: NoteRow::generateRepeats(ModelStackWithNoteRow, unsigned long, unsigned long, long, Action) at /src/deluge/model/note/note_row.cpp:2936
0x200d0e51: NoteRow::generateRepeats(ModelStackWithNoteRow, unsigned long, unsigned long, long, Action) at /src/deluge/model/note/note_row.cpp:2867 (discriminator 1)

0x200cea85: NoteRow::setDrum(Drum, Kit, ModelStackWithNoteRow, InstrumentClip, ParamManager, bool) at ??:?
0x20100219: Sample::percCacheClusterStolen(Cluster) at /src/deluge/model/sample/sample.cpp:903
 (inlined by) Cluster::steal(char const) at /src/deluge/storage/cluster/cluster.cpp:222
0x200d0dcf: NoteRow::generateRepeats(ModelStackWithNoteRow, unsigned long, unsigned long, long, Action) at /src/deluge/model/note/note_row.cpp:2944
0x200d0e19: NoteRow::generateRepeats(ModelStackWithNoteRow, unsigned long, unsigned long, long, Action*) at /src/deluge/model/note/note_row.cpp:2881

m-m-adams, 2025-03-29:
Could you check again? The last commit fixed a lot of crashes I was getting in the nightly

soymonitus, 2025-03-29:
Fixed with latest change! finally 1.3 beta is stable enough to use it daily 💃

closed 2025-03-29 by soymonitus