SynthstromAudible/DelugeFirmware #3512: Sample Playback Artifacts When Using as MIDI Slave

vanBuffelen, 2025-03-24:
### Please describe the problem:

When using the Deluge as a MIDI slave (via a DIN cable), artifacts appear when playing samples. I have tested it one-on-one with a Digitakt, Hydrasynth, and DAW as the master—each time with the same result. I have also tried using different MIDI cables, but the issue persists. It seems that something is going wrong with syncing the Deluge’s MIDI clock.

https://github.com/user-attachments/assets/0871dadf-0346-413a-9288-9e27febd3992

### What is the expected behavior?

no artifacts

### When did you start noticing the issue?

a few weeks ago

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly, other

### What is the firmware name:

1.2 and 1.3 nightly

davidvandenbor, 2025-04-16:
I also have sync issues when syncing another devices to the Deluge, but I have also noticed this when syncing 2 Deluge machines together! They sync, but the tempo readout on the Deluge oled screen doesn't match the tempo send out by the other Deluge. Deluge 1 song plays at 111 BMP. But the oled screen on Deluge 2 reads 115 BMP! They SOUND like they play in sync... except when you STOP Deluge 1, you can hear by the way the delay of Deluge 2 bleeds out, that “something's” off. Because the delay sound warbled and out of sync!

davidvandenbor, 2025-04-16:
I'm using Aprils nightly: deluge-v1_3_0-nightly+2025_04_02-cc9091b.bin

seangoodvibes, 2025-05-15:
@davidvandenbor can you confirm if this is still an issue for you with 1.2.1?

Thanks