SynthstromAudible/DelugeFirmware #3518: Midi recording from grid view creates ghost notes on first beat

MarW1N, 2025-03-26:
### Please describe the problem:

Recording midi using external keyboard on the grid view creates ghost notes on the first beat of the first bar. They are most often 0 velocity.

This is an issue that i first noticed in early januari and has been discussed in the nightly testing channel but I now formally report it so that it doesn't get forgotten. To replicate, all you need is a midi keyboard connected.

### What is the expected behavior?

No midi notes recorded onto the track other than those sent by the keyboard.

### When did you start noticing the issue?

Around new year 2024-2025

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.3.0-NIGHTLY-AABD8A9

### If possible provide the steps to reproduce the issue and upload additional media:

The procedure to replicate the problem is as follows. Make sure that you have set arm+record on new grid buttons.

1. Create a new song.
2. Go to grid mode.
3. Enable metronome (for timing purposes)
4. Select green mode
5. Press Play
6. Press Record
7. Press empty grid on first column to start recording.
8. Record a note, sometime during first beat, but leave a 8th note silent at least to easily display the ghost notes.
9. End recording by pressing the same grid button again
10. If you check the track, only the note you pressed should be there and be played back in a loop.
11. Repeat steps 6-9
12. Now a silent, or a very short, ghost note (or sometimes multiple notes) is played back directly on beat one and is shown when looking at the track.
13. Repeat as many new recordings as you like. For every new track made on that instrument these ghost notes will be recorded.
14. Now, delete the last recorded track by holding the grid key and pressing delete.
15. Repeat steps 6-9.
16. The last recording will be as intended without ghost notes.
17. Repeat steps 6-9.
18. Ghost notes are back in the recording.

If you create a new instrument, the ghost notes will not be recorded for the first track on that instrument, but the following recordings will have ghost notes. As soon as the last recorded track is removed from a grid column, new recording will work.

If you go back to the first instrument and record a new track while the "last recorded track" is present, recording will produce ghost notes. So the problem persists for each column in the grid until the last recording is removed.

MarW1N, 2025-07-12:
This is what I have found while looking into it a bit more.

deluge-v1_3_0-nightly+2024_12_21-6f48502.bin - seems to be working properly, no ghost note recorded when testing
deluge-v1_3_0-nightly+2024_12_25-0065a7f.bin - creates one ghost note when i tested, velocity 0, but I could still hear a short sound.

seangoodvibes, 2026-02-10:
Hi can you please confirm if this has been resolved with the latest nightly build?

Thanks

MarW1N, 2026-02-12:
> Hi can you please confirm if this has been resolved with the latest nightly build?

I still get ONE silent note in the beginning. But tried several times and didn't get a whole bunch of them at least.

closed 2026-04-06 by seangoodvibes