SynthstromAudible/DelugeFirmware #3520: CV tracks: gate timing problems when pressing UI grid buttons

therealstaubi, 2025-03-26:
### Please describe the problem:

Timing of gate impulses in CV clips is massively impacted by pressing UI buttons in grid mode. When unmuting a clip the displayed countdown in thew first row also seems to impact CV gate timing.

### What is the expected behavior?

UI interaction does not impact gate timing

### When did you start noticing the issue?

Observed in deluge-v1_3_0-nightly+2025_03_25-541d2dc and nightlies before

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_03_25-541d2dc

### If possible provide the steps to reproduce the issue and upload additional media:

Setup: Create a song in grid view and populate some columns with clips. Add a CV clip on channel 1and evenly fill it with notes (e.g. eights). Connect an external synth which is triggered by the gate 1 output (in my case Wirehead instruments Freaq FM)
Start sequencer and unmute CV 1 clip. Go back to grid view and randomly mute/unmute other clips --> you will hear that the external synth plays unevenly until the muted/unmuted clip has switchted its state.

therealstaubi, 2025-03-26:
The issue has also been present in nightlies before. Midi timing is not affected.

seangoodvibes, 2025-05-15:
@therealstaubi can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

pssssssst, 2025-05-24:
> [@therealstaubi](https://github.com/therealstaubi) can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

With todays beta (May 24) and 1.2.1 it's still is an issue.
also sending MIDI CCs from the deluge messes with the gate timing in the same way.

therealstaubi, 2025-05-25:
I tested the last firmware releases. Find the results here:
deluge-c1_0_1.bin --> no timing issues on gate 1
deluge-c1_1_1.bin --> no timing issues on gate 1
deluge-c1_2_0.bin --> timing issues on gate 1, especially while a clip of the song clip is activated but not yet playing (pad blinking)
deluge-c1_2_1.bin --> timing issues on gate 1, especially while a clip of the song clip is activated but not yet playing (pad blinking)
deluge-v1_3_0-beta+2025_05_25-9eb9213.bin --> timing issues on gate 1, especially while a clip of the song clip is activated but not yet playing (pad blinking), it seems to behave slightly different than with fw deluge-c1_2_1.bin but timing is still notably affected.

So the timing issues are present since 1.2.0

BenUniqcode, 2026-02-03:
I just ran into this one. Just wanted to point out that (for me anyway) it's not only Grid Mode, it's the same in Rows Mode. Just press any clip's launch pad (doesn't matter whether that clip is Synth, Kit, CV... I haven't tried Midi but I imagine it's the same), and the gate timing on any playing CV clips gets totally messed up until the launched/muted clip actually starts or stops; then it goes back to normal.

I think I read something about the countdown using excessive CPU, maybe that's the reason? 

The OLED horizontal positioning is also very prone to get messed up while this is happening - that happens in lots of other places too but it's especially noticeable here. It's as if the screen is being updated way more often than it needs to be, and that's hammering the CPU so it can't keep up with the gate pulses. But timing of other clips doesn't seem to be affected. Do CV tracks have a lower priority than everything else?

m-m-adams, 2026-04-19:
This might be fixed by #4424, could you check again please? Thanks!