SynthstromAudible/DelugeFirmware #3522: Export tracks record silence instead of variations when tracks have notes with "of X" probability.

stephenterhune, 2025-03-27:
### Please describe the problem:

Exported tracks records audio for he first variation of the track when notes in the track have "1 of 2", "1 of 3", "2 of 2", etc. probability. The remainder of the length of the track is recorded as silence, seemingly indicating that the "of X" notes influence the recorded length, but that there are no notes to record beyond the first variation.

I would not have expected anything beyond 1x through the track with percent probability, but it seems like we should expect the "of X" probability features to record audio for all of the "X" times through the track.

### What is the expected behavior?

The exported audio for each track should either include all of the "of X" probability variations (through X), or should be cropped to include only the first variation. The export should not be padded with silence.

### When did you start noticing the issue?

Just now, but this is the first time I've tried to do this kind of export. I can imagine this is a weird thing to try to do, but it's also weird output behavior. Just wanted to put it out there.

### Is there a relevant Pull request?

Not that I know of.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

c 1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

Create a synth track and edit the track to add steps. My scenario included 2 bars. At least one of the steps should be set to "1 of 2" probability, and other set to "2 of 2". Go back to song view and use the save + record combo to export the tracks. Observe the tracks on the SD card and play them back on a computer. In my experience, this results in silence for the last half of the exported audio.

BTW I will be very interested to find out if this is user error of some kind. I so, thank you for looking.

seangoodvibes, 2025-03-27:
@stephenterhune turn off "export to silence" in the stem export menu if you do not want silence to be recorded afterwards. The export to silence feature is there to record reverb and delay tails.

It is not possible to extend the recording to include all possible iterations of notes.

stephenterhune, 2025-03-27:
You're right. Thank you.

closed 2025-03-27 by stephenterhune