SynthstromAudible/DelugeFirmware #3541: Samples loaded to kits have random pitch and level. Deluge alters it randomly upon loading them.

matreve, 2025-04-01:
### Please describe the problem:

When loading samples to a new kit, Deluge loads them at random pitch/speed and level (likely different level of normalization). This is easy to notice when loading the same sample to several rows - every time it is loaded it's loaded at slightly different pitch and level (typically quite low level that is very different from the one heard during the preview). 

### What is the expected behavior?

When samples are loaded into a kit they should have level consistent with the original WAV file (and its preview), consistent pitch matching that of the original WAV file, no matter what row they are loaded into, every time. 

### When did you start noticing the issue?

I started noticing it after downloading nightly"deluge-v1_3_0-nightly+2025_04_01-e56d559". 

### Is there a relevant Pull request?

not aware of

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_04_01-e56d559

### If possible provide the steps to reproduce the issue and upload additional media:

Create a new kit. Load any sample into any row by pressing its AUDITION pad + LOAD. Select a sample. Repeat the process for another row. For ease of comparison load the same sample. The two samples will have inconsistent pitch/speed and different levels (in some cases very low). Repeating the process will reveal randomness of how the samples are loaded.

soymonitus, 2025-04-08:
It is super easy to reproduce it (latest nightly)

m-m-adams, 2025-04-10:
Do we know what commit it started on?

PR #3582 (2025-04-10): initialize everything
Fix #3541 

Initializing monophonic expression values fixes the bug, but also initialize all the other things in voice that clang tidy is warning about just in case
PR #3583 (2025-04-11): initialize everything
Fix #3541 

Initializing monophonic expression values fixes the bug, but also initialize all the other things in voice that clang tidy is warning about just in case
closed 2025-04-10 by soymonitus