SynthstromAudible/DelugeFirmware #3542: cracks and pops on the latest Nightly Built

Bagatour, 2025-04-02:
### Please describe the problem:

some serious cracks and pops appeared on some of my previously recorded synth tracks in the latest Nightly Built. 

### What is the expected behavior?

there was no pops and cracks on the previous nightly

### When did you start noticing the issue?

when I switch to the latest nightly build

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

Bagatour, 2025-04-02:
and I also can't make the sidebar controls momentary. (velocity and mod wheel) They are both set to momentary on the defaults / UI / keyboard  menu but somehow they are always permanent.

m-m-adams, 2025-04-06:
Fixed by #3564 and #3560

closed 2025-04-06 by m-m-adams