SynthstromAudible/DelugeFirmware #3549: Endless note playback for synth with looped sample osc and monophonic polyphony

lburygin, 2025-04-05:
### Please describe the problem:

For synth patch that has:
- env1 sustain > 0
- sample as oscillator
- sample repeat mode set to loop
- polyphony set to monophonic
if I play in legato I will get endless playback that can be stopped only by:
- rebooting the Deluge
- setting the env1 sustain to 0

Seems like the synth engine getting stuck at the sustain phase in this case

### What is the expected behavior?

As soon as note released synth engine should proceed to release stage

### When did you start noticing the issue?

Just now

### Is there a relevant Pull request?

No as far as I know

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-E2EDB15

### If possible provide the steps to reproduce the issue and upload additional media:

1. make new synth clip, init patch
2. choose any sample as osc
3. set sample mode to loop
4. set polyphony to monophonic
5. press some audition pad and without releasing the pad, press some another audition pad
6. now if you release both audition pads, playback will be endless

seangoodvibes, 2025-04-17:
@stellar-aria re opening this issue due to discord comment:

<@300067623167852554> #3613 PR fixes #3549 issue with osc type = sample but the same issue arises now with osc type = sine | square | saw

(just for information)

lburygin, 2025-04-18:
So the updated case is now:

make a new synth clip, init patch

set oscillator type to sine, saw or square

set polyphony to monophonic

press some audition pad and without releasing the pad, press some another audition pad

now if you release both audition pads, playback will be endless

PR #3609 (2025-04-16): Replace conditional any_of with none_of
Fixes #3549, fixes #3601
PR #3613 (2025-04-16): Replace conditional any_of with none_of
Fixes #3549, fixes #3601
PR #3658 (2025-05-01): Fix stuck sustain stage in monophonic mode
Fixes #3549, #3601
Reverts #3609 PR - the previous any_of was correct
closed 2025-05-01 by seangoodvibes