SynthstromAudible/DelugeFirmware #3571: Crash while saving

Rudement, 2025-04-08:
### Please describe the problem:

1.3, pressed stop before trying to save. 

[image]

### What is the expected behavior?

To save

### When did you start noticing the issue?

Just now. Software is a couple of weeks old I think.  

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

-8C243F4

soymonitus, 2025-04-08:
Have you tried with latest nightly or beta?

Rudement, 2025-04-09:
> No, last night there was an error on the download page.

Saved it to the recent build while it was  playing. It saved without crashing 👍🏾 Some audio drop outs while saving, but it was a processor heavy Song (that’s why I stopped play the first  time I tried to Save). Thanks as always!This has been an awesome experience since 2019 and you’re anll adding to this already amazing machine! I hope this leads you to great futures!-Cheers