SynthstromAudible/DelugeFirmware #3575: Resampling Drops Notes

TheHypnotist333, 2025-04-08:
### Please describe the problem:

When I try to resample melodies notes get dropped. It only happens when there are multiple notes triggered at the beginning of the measure while there is recorded automation. This happens even on the simplest of synth patches and kits.

### What is the expected behavior?

The expected behavior is of course to resample clips without notes dropping.

### When did you start noticing the issue?

I've been having this issue for many months.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

This bug happens to me every time I'm working on a song but when I try to reproduce it from a blank song I can't figure out how to make it happen 100% of the time. If you open up a synth patch and record filter movement over a chord that begins at the beginning of the measure about 50% of the time notes in that chord will be dropped when you initiate the resample function.

seangoodvibes, 2025-04-08:
Does this occur on the 1.1 release?

TheHypnotist333, 2025-04-08:
No, it doesn't occur on the 1.1 release.

seangoodvibes, 2025-04-08:
> No, it doesn't occur on the 1.1 release.

Can you confirm if it’s been fixed in the 1.3 beta build

TheHypnotist333, 2025-04-08:
The problem has not been fixed on today's 1.3. It's extremely unpredictable in general, I had to trigger the resample about 7 times before a note dropped and then 3 tries later it happened again but then I tried something like 20 times and it never happen again. For this reason I'm not even 100% positive the issue doesn't exist in 1.1 because while I tried it quite a number of times maybe I happened not to get the error that time.
All I'm doing to create this issue now is loading a default synth and placing a 4 note chord that spans the screen. Then I hit record and press play and fiddle with the cutoff, pan, and volume knobs while it's looping to record automation. Then I stop playback and trigger the resampling (record+play); if I hear all notes play I stop playback and try to trigger the resampling again and again until it drops notes.

soymonitus, 2025-04-08:
I have also experienced this with resampling missing the first note. Not sure if I would be able to reproduce it now. It happened to me a month ago

seangoodvibes, 2025-05-15:
@TheHypnotist333 can you please confirm if this is still an issue for you with 1.2.1?

Thanks

TheHypnotist333, 2025-05-16:
It seems that resampling is working again in 1.2.1 but I'm definitely still getting clicking/glitching with simple automated patches

seangoodvibes, 2025-05-16:
> It seems that resampling is working again in 1.2.1 but I'm definitely still getting clicking/glitching with simple automated patches

What kind of automation are you using? I think you're getting a different issue now that's probably always been there. If you could open a new issue for that with more info that would be great. I'll close this issue as it was resolved.

Thanks

closed 2025-05-16 by seangoodvibes