SynthstromAudible/DelugeFirmware #3578: Short glitchy sound when currently playing and scheduling next song load (1.3 alpha)

lburygin, 2025-04-09:
### Please describe the problem:

If you play the attached song and while playing, press LOAD and choose the same song as currently playing, you will hear a short but very noticeable glitchy stutter-ish sound

For some reason it occurs on stable basis only with this song for me (at the moment)

### What is the expected behavior?

When scheduling next song to load, there shouldn't be any audio artifacts

### When did you start noticing the issue?

Since 1.3 alpha, doesn't reproduce on the 1.2 chopin

### Is there a relevant Pull request?

No AFAIK

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-72AA06F

### If possible provide the steps to reproduce the issue and upload additional media:

1. load the attached song
2. press play
3. press load
4. choose the attached song from the list and press load (or select)

[glitchy loading.zip](https://github.com/user-attachments/files/19665284/glitchy.loading.zip)

lburygin, 2025-04-10:
Reproduced this on other CPU-heavy songs. So a more general case is:
- make a very CPU heavy song, you can use BOD 6-7 patches to quickly eat all CPU
- save the song
- play the song
- press load
- choose the song from the menu and press load

seangoodvibes, 2025-05-11:
I think you're probably hearing the initialize of the grain FX in that song. It's a known bug atm. If you change the mod fx type on "SYNT3" to something other than Grain FX I think it should load ok.

Unfortunately there's no solution for this yet. As @m-m-adams explained on discord:

"it’s the allocation time for the 4mb buffer, if it has to steal clusters it takes too long and causes the beep

one solution might be to use @stellar_aria’s object pool for the buffer and keep one pre allocated"

lburygin, 2025-06-11:
Yeah confirmed, the grain is the cause of this buzzy sound

seangoodvibes, 2026-01-05:
This PR fixed this buzz sound in nightly: https://github.com/SynthstromAudible/DelugeFirmware/pull/3962

Should be able to port the fix to beta.