SynthstromAudible/DelugeFirmware #3587: External clock not passed through correctly on song load

wtkdwc, 2025-04-12:
### Please describe the problem:

when clocked externally via midi in, the clock on midi out randomly stutters when loading another song. tried using midi monitor and it seems that on press of the load button deluge sends out multiple clock pulses at the same time. this causes my elektron rytm to stop on song load, which is unusable for live playing of multiple songs.

clock messages go like this- normal clock
16:47:15.562    From ES-9 MIDI In    Clock
16:47:15.822    From ES-9 MIDI In    Clock
--    load pressed 
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
16:47:15.833    From ES-9 MIDI In    Clock
--    then normal clock
16:47:15.842    From ES-9 MIDI In    Clock
16:47:15.861    From ES-9 MIDI In    Clock

another song:
normal clock
16:52:06.318    From ES-9 MIDI In    Clock
16:52:06.336    From ES-9 MIDI In    Clock
16:52:06.355    From ES-9 MIDI In    Clock
--    song load
16:52:06.469    From ES-9 MIDI In    Clock
16:52:06.469    From ES-9 MIDI In    Clock
16:52:06.469    From ES-9 MIDI In    Clock
16:52:06.470    From ES-9 MIDI In    Clock
16:52:06.470    From ES-9 MIDI In    Clock
16:52:06.470    From ES-9 MIDI In    Clock
--    then normal clock
16:52:06.486    From ES-9 MIDI In    Clock
16:52:06.505    From ES-9 MIDI In    Clock
16:52:06.532    From ES-9 MIDI In    Clock

### What is the expected behavior?

clock should pass through unchanged regardless if changing song or not

### When did you start noticing the issue?

current nightly (7.4.), tested on community 1.1 and it does not happen there

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

8e4f1ab

wtkdwc, 2025-04-17:
also noticed when saving song, the tempo gets all over the place for a while, (like from 140 to 300 briefly and then back)  this can be seen even on the deluge display

seangoodvibes, 2025-05-27:
@wtkdwc sorry for responding so late, but I see you were getting this on the nightly

The nightly is very bugged so we're not advising anyone to use it.

Have you tried 1.3 beta?

Thanks

wtkdwc, 2025-05-27:
@seangoodvibes yes issue persist on todays 1.3 beta build

wtkdwc, 2025-06-24:
I tried to swap rytm for digitakt, and it doesn’t stop itself on deluge song load, but jumps to double speed for a while too, resulting in running out of sync. I actually made a short video of it, here I load two songs on deluge, pay attention to the tempo in corner https://youtube.com/shorts/60wFlbszt08?si=Zn4yVALHHh1V_eR8