SynthstromAudible/DelugeFirmware #3640: Crash when MIDI learning clip master level

arealhumanbeing, 2025-04-22:
### Please describe the problem:

Using a Michigan Synth Works XVI-M faderbank:

1- Open Kit Clip
2- Hold SHIFT and press MASTER LEVEL shortcut
3- Hold LEARN/INPUT
4- Move fader on the XVI-M.

The screen then shows "LEARNED", but freezes.

[image]

### What is the expected behavior?

For that clip's MASTER LEVEL to be controllable by an external MIDI fader.

### When did you start noticing the issue?

Today, 4/22/25 at 3:00pm PST.

### Is there a relevant Pull request?

I don't know.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

Chopin 1.2

### If possible provide the steps to reproduce the issue and upload additional media:

Same thing occured on another Kit Clip. It worked with a Synth Clip, however.

arealhumanbeing, 2025-04-22:
It works with Audio Clips as well, so it only impacts Kits.

seangoodvibes, 2025-04-23:
> It works with Audio Clips as well, so it only impacts Kits.

Can you confirm if you have this issue on 1.1 or 1.3?

arealhumanbeing, 2025-04-26:
Sorry for the delay in response here. 1.3

arealhumanbeing, 2025-04-26:
I have not tried to do this with either 1.1 or 1.3 yet. I will see if I can test it this weekend.