SynthstromAudible/DelugeFirmware #3644: Select horizontal menus with synth-kit-midi-cv buttons doesn't work for envelope 1-4 menus (1.3 alpha)

lburygin, 2025-04-26:
### Please describe the problem:

Recently added ability to select horz menu item using synth-kit-midi-cv buttons works for all menus except envelope 1-4 menus. If you enter envelope menu and press any of synth-kit-midi-cv buttons, it does nothing.

### What is the expected behavior?

This feature should work consistently for all horz menus

### When did you start noticing the issue?

Today, on the latest alpha

### Is there a relevant Pull request?

No AFAIK

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-E222803

### If possible provide the steps to reproduce the issue and upload additional media:

1. enable horizontal menus feature in community settings
2. make a new song
3. make a new synth clip
4. enter the synth clip
5. enter env1 menu using the env1 attack shortcut
6. try to change menu item by pressing kit/midi/cv button

seangoodvibes, 2025-04-26:
Seems like it's because the envelope menus weren't setup properly as horizontal menus

If someone fixes that then the synth / kit / midi / cv buttons should start working automatically 

The envelope menus seem to be the only ones not setup properly

lburygin, 2025-04-26:
@seangoodvibes 
Seems like it is more of inheritance problem, currently we have structure:

Submenu
├──  HorizontalMenu
└── SubmenuReferringToOneThing
           └──Envelope

So Envelope menus currently don't have all behaviours of HorizontalMenu.
Maybe we need some class that has traits both of SubmenuReferringToOneThing and HorizontalMenu

lburygin, 2025-04-26:
@seangoodvibes 
Fixed in related PR, may I ask to review it please

PR #3645 (2025-04-26): Fix horz menu selection using synth-kit-cv-midi buttons for envelopes
Fixes #3644
closed 2025-04-26 by seangoodvibes