SynthstromAudible/DelugeFirmware #3646: Frequency Modulation (FM) broken in nightly

seangoodvibes, 2025-04-26:
### Please describe the problem:

Loaded up latest nightly build and tested my FM kit and all the sounds were completely distorted. Sounded like there was a ton of noise distortion added.

Also happens with the HPF if you turn FM up to max.

### What is the expected behavior?

FM works as before

### When did you start noticing the issue?

Today

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

seangoodvibes, 2025-05-11:
Confirmed as of today that this bug is still present in nightly. So the other audio fixes that went didn't address this one.

m-m-adams, 2025-06-20:
Rendering sine waves in general is broken in nightly, bisects to same commit

PR #3852 (2025-06-20): Bugfix/revert argon upgrade
Revert #3612 

I've ran through this code a few times this week and can't find where the bug comes from. Reverting for now so that nightly is usable again 

Fix #3646 
Fix #3557 
Fix #3551 
Fix #3577 
Fix #3742 

I'll open a second PR with the same fix for beta since it doesn't need argon reverted first
closed 2025-06-20 by seangoodvibes