SynthstromAudible/DelugeFirmware #3650: Sample based synth preset sounds correct on 1.3 when broken on 1.2

abstractrealism, 2025-04-28:
### Please describe the problem:

Weird issue, but documenting it just in case it helps solve anything else. 
During the recent patch competition, one of the presets I submitted had a separate single-cycle waveform for each oscillator. Someone converted the samples for all the presets to 44.1khz. On 1.2 this broke the sound of my preset as the loop points were now off. On 1.3 though it still sounded fine, as if it was still looping correctly. 

### What is the expected behavior?

I think probably 1.3 should match 1.2? In this case it sounded correct on 1.3 and not 1.2 but maybe there will be cases when the not-matching is a problem? Not sure, feel free to close this if this is a non-issue, but @seangoodvibes seemed surprised to hear about the difference so I figured I'd document it to be safe. 

### When did you start noticing the issue?

Tested both a recent 1.3 alpha (last couple days) and the one I'd been on from mid-March.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### If possible provide the steps to reproduce the issue and upload additional media:

Create sample-based preset. Set loop points. Change the samples it's using from 48khz to 44.1khz. On 1.2 the preset will now sound off. On 1.3 it will still sound the same. (See attachment for an example, zip contains the preset I submitted which sounds correct on 1.2 and 1.3, and the preset as included in the community preset pack draft which sounded the same on 1.3 but wrong on 1.2)

[problem preset.zip](https://github.com/user-attachments/files/19932859/problem.preset.zip)

seangoodvibes, 2025-05-15:
@abstractrealism can you confirm if this is still an issue on 1.2.1?

Thanks

abstractrealism, 2025-05-29:
> [@abstractrealism](https://github.com/abstractrealism) can you confirm if this is still an issue on 1.2.1?
> 
> Thanks

Apologies for the delay, I'll try and test that this week, however to be clear the 1.2 behavior was what was expected, 1.3 sounded correct despite the sample rate changing, which was unexpected.