SynthstromAudible/DelugeFirmware #3652: Odd envelope interaction during another track playback

wyvc, 2025-04-29:
### Please describe the problem:

Maybe a super niche bug (or behavior), but I'm reporting it as it theoretically could have implications on synth logic. Tested on 1.3, 1.2, and official, only on official the behavior is not present. Hard to describe but I have audio samples. While using envelope to modulate transpose, and then using an envelope (same or other) to modulate envelope (decay), and when envelope attack is set to 0, if a voice from another track is playing, the starting amount of transpose is at a different level during interaction with the other track.

### What is the expected behavior?

Voices from track 2 wouldn't influence the synth logic from track 1.

### When did you start noticing the issue?

Recent, but tested on 1.2, and 1.3, both behave the same.

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta, Release 1.2/Chopin

### What is the firmware name:

deluge-v1_3_0-beta+2025_04_27-d1ac415 and1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

Step by step:
- Turn on Deluge
- New Song
- Enter Track 1
- Create new Init Synth
- Modulate Master Transpose with Envelope 2 -> 30
- Set Envelope 2 Attack -> 0
- Adjust Envelope Decay -> 10 _(behavior is more pronounced at this value for some reason)_
- Modulate Envelope 2 Decay with Envelope 2 -> 30 _(or another envelope, i.e. similar behavior using Env3 and Env4 instead of Env2 here)_
- Apply note C4 at all 16 steps _(chosen note doesn't matter)_
- Press Play and note that voice sounds as expected
- Enter Grid/Row mode and Create new Track 2
- Enter Track 2
- Create new Init Synth _(or use any preset but it's easier to diagnose with Init)_
- Apply note C3 at 1, 8, 9 _(to test, or at random)_
- Reduce volume of Voice 2 to observe affect on Voice 1
- Press Play and note that the voice from Track 1 transpose value is changed during playback of voice from Track 2
- Repeat process on 1.3, 1.2, and Official, and note that behavior only occurs on community firmware.

[test.zip](https://github.com/user-attachments/files/19967502/test.zip)
https://github.com/user-attachments/assets/998e288c-8983-4fb5-8b63-7640eb65f954
https://github.com/user-attachments/assets/48d67d69-556a-4b74-b11f-3ce8b4fffa78

m-m-adams, 2025-04-30:
I'm not able to reproduce this consistently on nightly (well nightly from a week ago since current nightly is broken). I think what's happening is just a difference in audio frame size that's really audible for pitch rendering with envelopes modulating themselves. The deluge render frame size is dynamic and increases at higher load - official always keeps it as low as possible, the community builds aim for consistency between frame. This is almost always better since it makes voice dropping rarer and more predictable but longer render frames will make the envelope modulating itself less exponential, resulting in a higher pitch

m-m-adams, 2025-04-30:
the tl;dr is don't do this but we should add an extra control to make envelopes more exponential without relying on the mod matrix