SynthstromAudible/DelugeFirmware #3653: Chord keyboard view doesn't respect max voices limit (1.3 alpha)

lburygin, 2025-04-29:
### Please describe the problem:

When playing via chord view or chord library view, count of concurrent voices can go beyond limit of max voices.
The usual isomorphic and other types of keyboard work fine.

### What is the expected behavior?

Max voices limit should be applied consistently no matter which view is used

### When did you start noticing the issue?

today, on the latest 1.3 alpha (doesn't reproduce on 1.2 chopin)

### Is there a relevant Pull request?

No AFAIK

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-D1AC415

### If possible provide the steps to reproduce the issue and upload additional media:

1. make a new song
2. make a new synth clip, init synth
3. set max voices count to 3
4. enter keyboard view
5. switch view type to chord/chord library
6. try to play a couple of chords simultaneously - you will be able to do it

m-m-adams, 2025-04-30:
That's deliberate, we made a choice to always allow them to start notes. The 8 voice limit you see them hitting is because that's the maximum number of pressed audition pads we can track

lburygin, 2025-04-30:
@m-m-adams 
Monkey see undocumented difference, monkey go open GitHub issue :))

Now thinking about it, makes sense, maybe it's a lil bit confusing - if you record some chords it will sound different from playing via chord view because the previous recorded chord will be cut by the next recorded chord

But I see now why it was changed, thanks for answering

lburygin, 2025-04-30:
For example I use a trick where max voices set to 3 and env1 release set to long (40-50), this allows 3-note-chords to be played with long release but instantly be interrupted with next chord, it sounds cool and prevents from dirty releases

lburygin, 2025-04-30:
> Oh for that you can use the "auto" polyphony, it's actually exactly what it does
Cool thank you! Didn't know about it
I'm closing issue then

closed 2025-04-30 by lburygin