SynthstromAudible/DelugeFirmware #3681: Glitchy audio clip alongside kit clip

epedroni, 2025-05-10:
### Please describe the problem:

I have a song with two clips: one kit clip and one audio clip recorded from the line input. If I mute the kit clip and play, the audio clip sounds normal. With the kit clip unmuted, the kit sounds normal but the audio is glitchy. The issue is also reproducible with the kit clip volume set to 0 as long as the clip is not muted.

### What is the expected behavior?

The audio clip should sound the same regardless of other clips being muted or not.

### When did you start noticing the issue?

Yesterday (2025-05-09) when I created the song. I did not observe the issue in other similar songs in the last few weeks.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

C1.2.0 (1c39a9f)

### If possible provide the steps to reproduce the issue and upload additional media:

1. Open attached song
2. Play with both tracks unmuted, audio sample is glitchy
3. Mute kit clip (even while playing), audio sample is no longer glitchy

[sample-song.zip](https://github.com/user-attachments/files/20140438/sample-song.zip)

seangoodvibes, 2025-05-10:
@epedroni 

I think the problem is that your kit is really big and you have I think 33 samples in it and they aren't the smallest samples either.

the deluge caches a small portion of the samples and streams the rest. but when it can't stream fast enough, it drops other samples to make room in the cache for other samples.

I think what you are experiencing is drop outs from hitting the deluge's memory limits.

--

also when I look at your samples, they are 24 bit which is the worst bitrate for the deluge.

the best bit rate for performance are 16 bit samples, then 32 bit samples. 

one explanation from our discord:

"Internal processing always happens at 32 bit. Converting 16 to 32 bit is just as fast as moving 32 bit samples around so it doesn’t add any overhead. Converting 24 bit to 32 bit is very slow because it 24 bit isn’t a native size on the cpu. It has to be done 8 bits at a time to read all the data correctly and ends up taking 3x as long as the others"

--

other culprits for poor sample performance:

"bigger culprits are samples that are not 44.1k and 16bit, long release times, use of multisamples, synths/sounds that could ostensibly be monophonic that are set to polyphonic that rack up voice count "

epedroni, 2025-05-11:
Thanks for the quick response! I changed the bit depth of all the samples to 16 and also made them all mono, the glitches are still there but definitely less noticeable. I also tried with a few of the factory kits, same thing. This is a bit disappointing because it really limits how well the Deluge can be used as a looper. A few follow-up questions:

* Was this limitation also in the stock firmware? I don't remember coming across it but it's been a while
* Is it possible to have real-time feedback on the OLED display that I am hitting a memory or CPU bottleneck?
* Would it mitigate the problem to use a faster SD card? This is with a [Samsung Pro Plus 256GB](https://www.samsung.com/us/computing/memory-storage/memory-cards/pro-plus-adapter-microsdxc-256gb-mb-md256sa-am/)

seangoodvibes, 2025-05-11:
> Thanks for the quick response! I changed the bit depth of all the samples to 16 and also made them all mono, the glitches are still there but definitely less noticeable. I also tried with a few of [...]

Try the canvas go plus sd full size sd card. It can more than double sample streaming performance compared to stock card. The improvement is very noticeable.

As for looping, you should be looping with audio clips and not kits.

I've personally had no issues looping with audio clips. It's my favourite deluge workflow.

If you're combining audio clips with a ton of long synth / kit clip samples, then you're really impacting the deluge's ability to handle samples.

You can also adjust the priority of your clips and sounds to decide what audio should be dropped first when you hit the deluges limits.

If you're using the 1.2 firmware, try 1.1.1 as it currently performs better than 1.2.

These issues were also present on the official firmware yes.

Basically if you need to loop long audio samples, it's preferred that you loop them in audio clips.

Kit clips perform best with short samples. I saw a lot of 5-7 second samples in your kit.

The deluge's original benchmarks for samples in kits were based off taking a short drum sample (like a clap). So if your samples are much longer than that, then you won't get to use as many samples in a kit before running into issues.

Also in terms of sample size:

"under 128kb should always be cached but there's no upper limit. If the song isn't using more than 64mb in total then the big ones will be cached as well"

"I think if the song is using tons of memory then only 64kb will always be cached but I'm not 100% on that"

So ideally if you want the best sample performance; you'd want your samples to be cached which will definitely happen if they're 64kb or smaller. Bigger samples can still be cached but if the deluge needs more memory than the bigger samples will be streamed which means the bottleneck will be the sd cards streaming speed. 

If you hit the sd cards streaming limits, then you start hitting issues.

This is why displaying info on cpu usage / memory usage is difficult because it's a complex calculation for which we do not have any accurate way of measuring it and also the accuracy may vary greatly based on your specific deluge use case.

epedroni, 2025-05-12:
> As for looping, you should be looping with audio clips and not kits.

So if I understand correctly, my options to do this are either the automated audio export or setting the kit clip as the input to an audio clip and bouncing it in real time. It would be cool if arranger tracks were automatically rendered for playback, that way I wouldn't have to re-render the clip it every time I change it. But maybe that's too slow to do in the background...

Anyway thanks again for the support!

seangoodvibes, 2025-05-12:
> So if I understand correctly, my options to do this are either the automated audio export or setting the kit clip as the input to an audio clip and bouncing it in real time. It would be cool if [...]

I don't think I'm understanding your workflow

I assume by audio looping you mean play an external instrument and send audio into the deluge for live looping

That is what audio clips are meant for

If you're trying to loop / sequence samples, then i recommend:

1) get an sd card with faster sample streaming - eg canvas go plus

2) use shorter samples, smaller file sizes

3) adjust the priority of your clips and kit sounds to decide which sounds should be dropped first when you hit clips

4) If you need to use render more complex sequences with kit samples, then yes you can bounce the audio to an audio clip. You don't need to use audio export for that. You can route the audio from a kit to an audio clip internally and live loop / record the audio and then mute your kit clip after. Note: I'm not sure that muting the kit clip fully frees up memory. I think it still caches. So if you really want to lighten your song you need to delete the kit clip after.

5) lastly, if you have a large kit and aren't using all the samples in that kit, it is recommended to delete the samples you're not using because those samples will continue to be cached even if they're not used

seangoodvibes, 2025-05-15:
@epedroni have you tried 1.2.1? Curious if it improves your situation

Thanks!

closed 2025-06-19 by m-m-adams