SynthstromAudible/DelugeFirmware #3690: Sample names not displaying on 7 seg

Gibbonnaboa, 2025-05-11:
### Please describe the problem:

 I downloaded the 1.3 beta a few days ago, it is 1.3.0-beta-d105dbc. 
 The problem I'm having and have been having with other versions now(sorry, should've spoke up) is that when I'm going through sample banks the 7 seg doesn't display the sample name on the display, I just get a 2 cursors chasing around the outside, clockwise, as if it's loading. I thought that as I'd read that there was a lot of under the hood speeding up of processes coming in this update it would have stopped with this beta, that's why I never mentioned it earlier. 
 I've filmed it, I'll send the footage. The whole time I've been writing it's still going round and round as if loading, it never gets there.
 This never used to happen, I can't remember specifically which update this started with. 

 Thanks. Adam. 

### What is the expected behavior?

To just display the sample name

### When did you start noticing the issue?

Since I started on 1.3, I think it was a nightly then not the beta but as I said earlier I thought the beta would fix it.... hey, I ain't a computer guy 🤷

### Is there a relevant Pull request?

?

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly, Beta

### What is the firmware name:

1.3.0-beta-d105dbc

### If possible provide the steps to reproduce the issue and upload additional media:

Just create a new kit, take off your headphones and try and find your favourite samples without listening. If you can't find em you've recreated it. 
 Could be an idea for some kind of involved randomisation to make weird kits though.

https://github.com/user-attachments/assets/a676a5a6-d69e-465b-bbdb-14c48a253548

Gibbonnaboa, 2026-10-21:
I'm still experiencing this bug. I am running a different memory card to when I first reported this issue if that info helps. 
It's irritating because I can't really see what I'm looking for. This has been made worse by the fact the last time I updated my 1.3 beta it reorganised all the files on my memory card, I used to have a good idea of where things were. I'm currently running 1.3.0-beta-1760e

https://github.com/user-attachments/assets/c2af6e0f-4463-4633-8489-638d5cec671b

Gibbonnaboa, 2026-10-21:
Sorry, I uploaded the wrong video! The issue is sporadic but if say happens more often than not.
Regards, adam

https://github.com/user-attachments/assets/0a6eb8ba-1e3c-45b7-8da1-810389ba8863

seangoodvibes, 2026-10-21:
> Sorry, I uploaded the wrong video! The issue is sporadic but if say happens more often than not.
> Regards, adam
> 
> https://github.com/user-attachments/assets/0a6eb8ba-1e3c-45b7-8da1-810389ba8863

Hi Adam, when was the last time you updated the firmware?

Could you try the latest rc2 firmware?

Thanks

Sean

Gibbonnaboa, 2026-10-21:
Hi Sean.
 I got straight on it, this is the rc2 firmware now. I added a video again to demonstrate it still happens, the sample plays but the name doesn't appear, only for a split second if it does.
 It seems to be in the bottom tier of the menu, so going through folders I'm pretty sure is okay, it's once you get to the actual samples in the folders the issue appears. 
 
 Appreciate all the good work man. 
 Thanks 
 Adam

https://github.com/user-attachments/assets/b7c2f2c5-92d5-4513-83e8-874ed5ba6cea

seangoodvibes, 2026-10-21:
> Hi Sean.
>  I got straight on it, this is the rc2 firmware now. I added a video again to demonstrate it still happens, the sample plays but the name doesn't appear, only for a split second if it [...]

Thanks for confirming it, we'll get this fixed

Cheers

closed 2026-10-21 by seangoodvibes

Gibbonnaboa, 2026-10-21:
You lot don't fuck about do you! 
Should that be fixed in the next update?
Good work.
Adam

seangoodvibes, 2026-10-21:
> You lot don't fuck about do you! 
> Should that be fixed in the next update?
> Good work.
> Adam

lol will let you know once we have a new firmware version published for you to test

Cheers!

seangoodvibes, 2026-10-21:
> 18\. You lot don't fuck about do you!
>      Should that be fixed in the next update?
>      Good work.
>      Adam

Can you try this build?

https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/39596922487/artifacts/12345088096

Gibbonnaboa, 2026-10-21:
Hi Sean. 
 That seems to have sorted it, that's the best it's worked in a long time now, good job 👍 

Thanks, Adam 

https://github.com/user-attachments/assets/38c8f232-38a4-45bf-91c3-3ed6c92bd666

Gibbonnaboa, 2026-10-23:
Hi.
 This is the first time it's crashed on that firmware you sent me, I took a pic of how it lit up. I don't really know the right place to send these too or if they are helpful. 
 Apart from that it's going really well, I've probably put in at least 10 hours, the thing where you press shift and all the functions light up is brilliant. 
 Regards
 Adam 

[image]

stellar-aria, 2026-10-23:
@Gibbonnaboa please open a new issue for any new problems now that this is solved. We probably also need a bit more explanation of what you were doing when that crash happened, as per the new issue template.