SynthstromAudible/DelugeFirmware #3707: Level parameter setting from midifollow.xml not being saved with songs

davidvandenbor, 2025-05-16:
### Please describe the problem:

In the midifollow.xml file there are 2 parameters that deal with volume levels of a clip, one is pre-FX and one is post-FX. When you turn the golden rotator, in the OLED display ```<volumePostFX>``` popup shows up as **"MASTER LEVEL"**, and when you turn a rotator on the Faderfox ```<volume>``` shows up as **"LEVEL"** in the OLED display. In the default midifollow XML file the parameters show up as:

```
  <volumePostFX>7</volumePostFX>
  <volume>7</volume>
```
I noticed that The "master level" (the first orange button with the top-left gold rotator) was unreachable for my midi controller (faderfox ec4) but the moment I renamed one of the CC's to another number, the Deluge was able to distinguish between the two, so that worked:

```
  <volumePostFX>7</volumePostFX>
  <volume>114</volume>
```
But here's where the problem occurs: when I save songs, the ```<volume>``` setting doesn't get saved with the songs! When I re-open songs, the ```<volume>``` CC level gets set back to its default level again (25) So during the editing or performance of the song, the ```<volume>``` levels retain their positions within each clip, but the moment you save the song, all the ```<volume>``` levels get reset back again at 25.

So in a nutshell, only ```<volumePostFX>``` gets saved with a song, but ```<volume>``` doesn't.

### What is the expected behavior?

```<volume>```of clips should be saved with a song, just as ```<volumePostFX>``` levels are.

### When did you start noticing the issue?

After updating to the latest nightly, Beta 6 (deluge-v1_3_0-beta+2025_05_14-b0e6c3d.bin)

### Is there a relevant Pull request?

Not that I know of, I searched the open issues

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-beta+2025_05_14-b0e6c3d.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Steps are provided above, it should be pretty easy to stumble on the issue.

seangoodvibes, 2025-05-16:
@davidvandenbor where did you get volume from? I don't recall that ever being in the midi follow xml, I don't know what that parameter is but I don't think you should be using it, I think you found a backdoor into a deluge parameter that was not meant to be publicly accessible

seangoodvibes, 2025-05-16:
Yea I double checked, you're not supposed to be controlling that param directly. That param is intended for patch cables, such as velocity, to increase volume.

We won't be added anything to save changes to that param.

We definitely did not intend for anyone to have midi control access to that param.

seangoodvibes, 2025-05-16:
If you need to control volume, please use volumePostFX, I don't understand why your controller can't adjust volume with that parameter. Something must be missing in your midi follow config.

seangoodvibes, 2025-05-16:
I will keep this open actually so that we can block access to that param

davidvandenbor, 2025-05-16:
Hi Sean! 

Thanks for replying! Weird, I just opened the midifollow.XML and those parameters were there to be used, including the ```<volume>``` parameter. I have no idea how it got there, because I can't find an official midifollow.xml in the repository to compare it with. One thing though, it still was nice to be able to control the volume of each clip, without having the reverbs and delays killed post-fx wise. Somehow muting the clips pre-FX sounds musically more natural 🙂

But I get it though, having way too many parameters can also make sound design way too complicated than it should be.
Regarding the XML file: this likely means that if I remove the ```<volume>``` parameter from the XML file, the CC7 will probably be “returned” to the master level and the CC conflict between **master level** and **level** will be resolved. That's probably why my controller couldn't adjust it.

I'll also reply to the other issue that I posted (the one about the tempo indicator on the OLED screen). I have downloaded the nightly BETA 6, and a couple of glitchy behaviours were miraculously resolved. This version immediately **feels** more stable, there is also less culling. Everything runs fine, except for one mayor problem that has now arisen: loading or auditioning presets results in freezes and crashes 70% of the time, which makes it almost impossible to load synth/kit presets, slowing down music making. But I'll open another issue for this if I don't find an equivalent report on this in the issues section. 

Kind regards and thanks in advance, David!

seangoodvibes, 2025-05-16:
> Hi Sean!
>
> Thanks for replying! Weird, I just opened the midifollow.XML and those parameters were there to be used, including the `<volume>` parameter. I have no idea how it got there, because I [...]

Hey David, thanks for responding. I didn't realize <volume> was in the midifollow.xml file. It may have been added by mistake recently. I'll talk to the person that added it.

Re nightly and song loading crashes, the nightly is actually quite broken right now, so I recommend using the 1.3 beta instead. Is that what you are using? It's the one from this link: https://github.com/SynthstromAudible/DelugeFirmware/releases/download/beta/beta.zip

Thx

davidvandenbor, 2025-05-16:
Below is a copy of my midifollow.XML 
I also have two parameters for pitch in them. Is that also weird?

```xml
<?xml version="1.0" encoding="UTF-8"?>
<defaults>
	<defaultCCMappings>
		<pitch>112</pitch>
		<pitchAdjust>3</pitchAdjust>
		<portamento>113</portamento>
		<volumePostFX>7</volumePostFX>
		<pan>10</pan>
		<oscAPitch>12</oscAPitch>
		<oscBPitch>13</oscBPitch>
		<modulator1Pitch>14</modulator1Pitch>
		<modulator2Pitch>15</modulator2Pitch>
		<modFXRate>16</modFXRate>
		<modFXFeedback>17</modFXFeedback>
		<modFXOffset>18</modFXOffset>
		<waveFold>19</waveFold>
		<stutterRate>20</stutterRate>
		<oscAVolume>21</oscAVolume>
		<oscAPhaseWidth>23</oscAPhaseWidth>
		<carrier1Feedback>24</carrier1Feedback>
		<oscAWavetablePosition>25</oscAWavetablePosition>
		<oscBVolume>26</oscBVolume>
		<compressorThreshold>27</compressorThreshold>
		<oscBPhaseWidth>28</oscBPhaseWidth>
		<carrier2Feedback>29</carrier2Feedback>
		<oscAWavetablePosition>30</oscAWavetablePosition>
		<reverseProbability>36</reverseProbability>
		<spreadVelocity>37</spreadVelocity>
		<spreadOctave>39</spreadOctave>
		<spreadGate>40</spreadGate>
		<noiseVolume>41</noiseVolume>
		<rhythm>42</rhythm>
		<sequenceLength>43</sequenceLength>
		<chordPolyphony>44</chordPolyphony>
		<ratchetAmount>45</ratchetAmount>
		<noteProbability>46</noteProbability>
		<bassProbability>47</bassProbability>
		<chordProbability>48</chordProbability>
		<ratchetProbability>49</ratchetProbability>
		<arpGate>50</arpGate>
		<arpRate>51</arpRate>
		<delayFeedback>52</delayFeedback>
		<delayRate>53</delayRate>
		<modulator1Volume>54</modulator1Volume>
		<modulator1Feedback>55</modulator1Feedback>
		<modulator2Volume>56</modulator2Volume>
		<modulator2Feedback>57</modulator2Feedback>
		<lfo1Rate>58</lfo1Rate>
		<lfo2Rate>59</lfo2Rate>
		<compressorShape>60</compressorShape>
		<sidechainCompressorVolume>61</sidechainCompressorVolume>
		<bitcrushAmount>62</bitcrushAmount>
		<sampleRateReduction>63</sampleRateReduction>
		<lpfMorph>70</lpfMorph>
		<lpfResonance>71</lpfResonance>
		<env1Release>72</env1Release>
		<env1Attack>73</env1Attack>
		<lpfFrequency>74</lpfFrequency>
		<env1Decay>75</env1Decay>
		<env1Sustain>76</env1Sustain>
		<env2Attack>77</env2Attack>
		<env2Decay>78</env2Decay>
		<env2Sustain>79</env2Sustain>
		<env2Release>80</env2Release>
		<hpfFrequency>81</hpfFrequency>
		<hpfResonance>82</hpfResonance>
		<hpfMorph>83</hpfMorph>
		<bassFreq>84</bassFreq>
		<trebleFreq>85</trebleFreq>
		<bass>86</bass>
		<treble>87</treble>
		<reverbAmount>91</reverbAmount>
		<modFXDepth>93</modFXDepth>
		<env3Attack>102</env3Attack>
		<env3Decay>103</env3Decay>
		<env3Sustain>104</env3Sustain>
		<env3Release>105</env3Release>
		<env4Attack>106</env4Attack>
		<env4Decay>107</env4Decay>
		<env4Sustain>108</env4Sustain>
		<env4Release>109</env4Release>
		<lfo3Rate>110</lfo3Rate>
		<lfo4Rate>111</lfo4Rate>
	</defaultCCMappings>
</defaults>
```

davidvandenbor, 2025-05-16:
Hi Sean! Wow, that was a fast reply! 😄 Thanks, I will switch to the firmware version you mentioned. Kind regards, David!

seangoodvibes, 2025-05-16:
> Below is a copy of my midifollow.XML I also have two parameters for pitch in them. Is that also weird?
> 
> <?xml version="1.0" encoding="UTF-8"?>
> <defaults>
> 	<defaultCCMappings>
> 		<pitch>112</pitch>
> 		<pitchAdjust>3</pitchAdjust>
> 		<portamento>113</portamento>
> 		<volumePostFX>7</volumePostFX>
> 		<pan>10</pan>
> 		<oscAPitch>12</oscAPitch>
> 		<oscBPitch>13</oscBPitch>
> 		<modulator1Pitch>14</modulator1Pitch>
> 		<modulator2Pitch>15</modulator2Pitch>
> 		<modFXRate>16</modFXRate>
> 		<modFXFeedback>17</modFXFeedback>
> 		<modFXOffset>18</modFXOffset>
> 		<waveFold>19</waveFold>
> 		<stutterRate>20</stutterRate>
> 		<oscAVolume>21</oscAVolume>
> 		<oscAPhaseWidth>23</oscAPhaseWidth>
> 		<carrier1Feedback>24</carrier1Feedback>
> 		<oscAWavetablePosition>25</oscAWavetablePosition>
> 		<oscBVolume>26</oscBVolume>
> 		<compressorThreshold>27</compressorThreshold>
> 		<oscBPhaseWidth>28</oscBPhaseWidth>
> 		<carrier2Feedback>29</carrier2Feedback>
> 		<oscAWavetablePosition>30</oscAWavetablePosition>
> 		<reverseProbability>36</reverseProbability>
> 		<spreadVelocity>37</spreadVelocity>
> 		<spreadOctave>39</spreadOctave>
> 		<spreadGate>40</spreadGate>
> 		<noiseVolume>41</noiseVolume>
> 		<rhythm>42</rhythm>
> 		<sequenceLength>43</sequenceLength>
> 		<chordPolyphony>44</chordPolyphony>
> 		<ratchetAmount>45</ratchetAmount>
> 		<noteProbability>46</noteProbability>
> 		<bassProbability>47</bassProbability>
> 		<chordProbability>48</chordProbability>
> 		<ratchetProbability>49</ratchetProbability>
> 		<arpGate>50</arpGate>
> 		<arpRate>51</arpRate>
> 		<delayFeedback>52</delayFeedback>
> 		<delayRate>53</delayRate>
> 		<modulator1Volume>54</modulator1Volume>
> 		<modulator1Feedback>55</modulator1Feedback>
> 		<modulator2Volume>56</modulator2Volume>
> 		<modulator2Feedback>57</modulator2Feedback>
> 		<lfo1Rate>58</lfo1Rate>
> 		<lfo2Rate>59</lfo2Rate>
> 		<compressorShape>60</compressorShape>
> 		<sidechainCompressorVolume>61</sidechainCompressorVolume>
> 		<bitcrushAmount>62</bitcrushAmount>
> 		<sampleRateReduction>63</sampleRateReduction>
> 		<lpfMorph>70</lpfMorph>
> 		<lpfResonance>71</lpfResonance>
> 		<env1Release>72</env1Release>
> 		<env1Attack>73</env1Attack>
> 		<lpfFrequency>74</lpfFrequency>
> 		<env1Decay>75</env1Decay>
> 		<env1Sustain>76</env1Sustain>
> 		<env2Attack>77</env2Attack>
> 		<env2Decay>78</env2Decay>
> 		<env2Sustain>79</env2Sustain>
> 		<env2Release>80</env2Release>
> 		<hpfFrequency>81</hpfFrequency>
> 		<hpfResonance>82</hpfResonance>
> 		<hpfMorph>83</hpfMorph>
> 		<bassFreq>84</bassFreq>
> 		<trebleFreq>85</trebleFreq>
> 		<bass>86</bass>
> 		<treble>87</treble>
> 		<reverbAmount>91</reverbAmount>
> 		<modFXDepth>93</modFXDepth>
> 		<env3Attack>102</env3Attack>
> 		<env3Decay>103</env3Decay>
> 		<env3Sustain>104</env3Sustain>
> 		<env3Release>105</env3Release>
> 		<env4Attack>106</env4Attack>
> 		<env4Decay>107</env4Decay>
> 		<env4Sustain>108</env4Sustain>
> 		<env4Release>109</env4Release>
> 		<lfo3Rate>110</lfo3Rate>
> 		<lfo4Rate>111</lfo4Rate>
> 	</defaultCCMappings>
> </defaults>

Oh can you try deleting the midifollow.xml and re-running the deluge? it will recreate the file.

I think maybe what is happening is you're getting a mismatch of old midi follow xml settings and new midi follow xml settings

davidvandenbor, 2025-05-18:
Hi Sean! Thanks for the tip, I did this, and this is the file that was created by default by the Beta firmware https://github.com/SynthstromAudible/DelugeFirmware/releases/download/beta/beta.zip

```xml
<?xml version="1.0" encoding="UTF-8"?>
<defaults>
	<defaultCCMappings>
		<pitch>3</pitch>
		<pitchAdjust>3</pitchAdjust>
		<portamento>5</portamento>
		<volumePostFX>7</volumePostFX>
		<volume>7</volume>
		<pan>10</pan>
		<oscAPitch>12</oscAPitch>
		<oscBPitch>13</oscBPitch>
		<modulator1Pitch>14</modulator1Pitch>
		<modulator2Pitch>15</modulator2Pitch>
		<modFXRate>16</modFXRate>
		<modFXFeedback>17</modFXFeedback>
		<modFXOffset>18</modFXOffset>
		<waveFold>19</waveFold>
		<stutterRate>20</stutterRate>
		<oscAVolume>21</oscAVolume>
		<oscAPhaseWidth>23</oscAPhaseWidth>
		<carrier1Feedback>24</carrier1Feedback>
		<oscAWavetablePosition>25</oscAWavetablePosition>
		<oscBVolume>26</oscBVolume>
		<compressorThreshold>27</compressorThreshold>
		<oscBPhaseWidth>28</oscBPhaseWidth>
		<carrier2Feedback>29</carrier2Feedback>
		<oscAWavetablePosition>30</oscAWavetablePosition>
		<reverseProbability>36</reverseProbability>
		<spreadVelocity>37</spreadVelocity>
		<spreadOctave>39</spreadOctave>
		<spreadGate>40</spreadGate>
		<noiseVolume>41</noiseVolume>
		<rhythm>42</rhythm>
		<sequenceLength>43</sequenceLength>
		<chordPolyphony>44</chordPolyphony>
		<ratchetAmount>45</ratchetAmount>
		<noteProbability>46</noteProbability>
		<bassProbability>47</bassProbability>
		<chordProbability>48</chordProbability>
		<ratchetProbability>49</ratchetProbability>
		<arpGate>50</arpGate>
		<arpRate>51</arpRate>
		<delayFeedback>52</delayFeedback>
		<delayRate>53</delayRate>
		<modulator1Volume>54</modulator1Volume>
		<modulator1Feedback>55</modulator1Feedback>
		<modulator2Volume>56</modulator2Volume>
		<modulator2Feedback>57</modulator2Feedback>
		<lfo1Rate>58</lfo1Rate>
		<lfo2Rate>59</lfo2Rate>
		<compressorShape>60</compressorShape>
		<sidechainCompressorVolume>61</sidechainCompressorVolume>
		<bitcrushAmount>62</bitcrushAmount>
		<sampleRateReduction>63</sampleRateReduction>
		<lpfMorph>70</lpfMorph>
		<lpfResonance>71</lpfResonance>
		<env1Release>72</env1Release>
		<env1Attack>73</env1Attack>
		<lpfFrequency>74</lpfFrequency>
		<env1Decay>75</env1Decay>
		<env1Sustain>76</env1Sustain>
		<env2Attack>77</env2Attack>
		<env2Decay>78</env2Decay>
		<env2Sustain>79</env2Sustain>
		<env2Release>80</env2Release>
		<hpfFrequency>81</hpfFrequency>
		<hpfResonance>82</hpfResonance>
		<hpfMorph>83</hpfMorph>
		<bassFreq>84</bassFreq>
		<trebleFreq>85</trebleFreq>
		<bass>86</bass>
		<treble>87</treble>
		<reverbAmount>91</reverbAmount>
		<modFXDepth>93</modFXDepth>
		<env3Attack>102</env3Attack>
		<env3Decay>103</env3Decay>
		<env3Sustain>104</env3Sustain>
		<env3Release>105</env3Release>
		<env4Attack>106</env4Attack>
		<env4Decay>107</env4Decay>
		<env4Sustain>108</env4Sustain>
		<env4Release>109</env4Release>
		<lfo3Rate>110</lfo3Rate>
		<lfo4Rate>111</lfo4Rate>
	</defaultCCMappings>
</defaults>
```

As you can see, the weird additions are still there, even in the latest version, so the problem is in the default XML that is generated by the firmware. As you can see, there are 2 parameters for pitch and 2 parameters for volume, which is confusing. They also share the same CC number (7 and 3) which causes CC collisions/conflicts when controlling the Deluge with midi. What I also noticed: portamento with CC5 wasn't working for me a week ago, I had to rename portamento to 112 for it to work.

Again, thanks for your assistance, and the relentless and amazing work on the community firmware versions!

seangoodvibes, 2025-05-18:
> Hi Sean! Thanks for the tip, I did this, and this is the file that was created by default by the Beta firmware https://github.com/SynthstromAudible/DelugeFirmware/releases/download/beta/beta.zip [...]

Hey thanks for sharing, that's definitely a bug. Will fix that!

davidvandenbor, 2025-05-18:
Also, just confirmed: ```<pitch>3</pitch>``` is not working either! I also needed to switch it to an unassigned CC number (114)
Also noticed (or maybe it's just me) when I'm in grid mode and want to transpose a complete song, pushing and turning the most left black encoder doesn't work (turn + rotate). Semi transpose in grid view when pressing a clip also doesn't work either (turn + shift + rotate)

closed 2025-05-18 by seangoodvibes

seangoodvibes, 2025-05-18:
@davidvandenbor the midi follow xml file issues should be fixed the next time the 1.3 beta is published (later today / tomorrow).

It won't added volume or pitchAdjust to the xml anymore

If you're having another issue with the master transpose feature, please open a separate issue.

Thanks!