SynthstromAudible/DelugeFirmware #3709: Automation Clicking

TheHypnotist333, 2025-05-16:
### Please describe the problem:

There is clicking/glitching when playing back automated synth patches.

### What is the expected behavior?

To automate patches cleanly.

### When did you start noticing the issue?

Quite a long time, I'm not sure exactly.

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Make a new synth (shift+synth)
2. Place a single note that spans the full length of the sequence
3. Press play and automate the filter and resonance randomly turning the gold knobs
4. Randomly automate delay amount with gold knob.

During step 4 you should notice clicking and as the sequence continues to loop it continues to click seemingly randomly, not necessarily looping with the sequence.

You might need to be somewhat aggressive with the automation for this to happen but still within the normal realm of use.

[Automation Glitch.zip](https://github.com/user-attachments/files/20259441/Automation.Glitch.zip)

seangoodvibes, 2025-05-16:
Actually the real reason is your delay sync setting

Change the delay sync to 1 bar, for example, and it goes away

TheHypnotist333, 2025-05-16:
It's not the clipping, I turn the master volume down and the clicks are still there.

The delay... but what if I want quarter note delay? It shouldn't be clicking no matter the delay setting. Right? 
This really happens to me so often in practice when I'm working on songs, whatever delay settings I'm choosing are intentional and honestly normal.

seangoodvibes, 2025-05-16:
> It's not the clipping, I turn the master volume down and the clicks are still there.
>
> The delay... but what if I want quarter note delay? It shouldn't be clicking no matter the delay setting. [...]

Idk, I don't think this is a bug but just how the deluge's fx are. They've always been this way.

I honestly can't reproduce the click with a brand new song, recording filter cut off and resonance and delay, and doing it aggressively like you say with a single drone note.

It's also possible you're getting the delay into feedback territory if you're going over 25 for even a split second

You can try the analog delay instead of digital delay also

There's really no other solution we can offer here

seangoodvibes, 2025-05-16:
Here's a song I just made that doesn't click. Same setup as your songs.

I was very aggressive with the gold knobs recording automation and could not hear a click.

[Automation Song Sean.XML.zip](https://github.com/user-attachments/files/20260106/Automation.Song.Sean.XML.zip)

TheHypnotist333, 2025-05-17:
This was always my complaint about the deluge since I got it back in 2017, that it is very clicky even when used as intended, you basically must accept your music is going to have pops and clicks. I had hoped someday the developers would add smoothing to the fx but of course idk how that works and now you're saying it's practically not possible. 
The deluge is so powerful for production but this quality of the clicking does it such a monumental disservice in my opinion. There are so many super amazing creative effects I've discovered myself on the deluge over the years and planning to teach more about them on my YouTube channel but the fact the deluge clicks left and right with fx means i cannot recommend it to my intended audience. I make hip hop with soft ambient melodies, the clicking really messes with the vibe and sounds unprofessional to my fellow rappers and audience.
Very depressing to learn this is unfixable. I'm not really sure what I'm doing with the deluge anymore.

Thank you for looking into it and letting me know.

Please lrt me know one thing, do you know if it might theoretically be possible to accomplish if a team was paid enough money to clean up the effects? Or do you suspect it's really not possible due to some limitations of the hardware?

seangoodvibes, 2025-05-17:
> This was always my complaint about the deluge since I got it back in 2017, that it is very clicky even when used as intended, you basically must accept your music is going to have pops and clicks. I [...]

I'll ask the team if they think the clicking could be fixed with smoothing. Will get back to you.

seangoodvibes, 2025-05-17:
> This was always my complaint about the deluge since I got it back in 2017, that it is very clicky even when used as intended, you basically must accept your music is going to have pops and clicks. I [...]

Answer I got was no, fixing the delay "clicks" won't happen.

They did say that there is smoothing if you use modulation (eg "You can modulate the delay rate with lfo1 without causing clicks")

You can also learn patch cables to then gold knobs so you can use that to tweak the modulation depth and record automation.

There was work going into creating a new delay but it's on hold for the moment, I don't know if that delay will also click.

But check out the modulation trick and let me know if that clicks for you.

I'm sorry the clicks have been frustrating for you and your style of music making. I can understand how annoying that would be.

The one thing I can say is that we do have intentions to continue improving the sound engine. But that's more long term than short term.

ok-reza, 2025-05-17:
after trying to recreate myself with the initial intructions unsuccesfully, i took a look at the provided .xml file examples.

the clicks are being caused when the delay amount reaches 0 during automation, when it reactivates at any amount afterwards, a click will be produced.

maybe a solution is related to not completely disengaging the delay if there's delay amount automation occurring in a clip? as that is what it seems like the clicking artifact is generating from (switching from off and on delay states)

seangoodvibes, 2025-05-17:
> after trying to recreate myself with the initial intructions unsuccesfully, i took a look at the provided .xml file examples.
>
> the clicks are being caused when the delay amount reaches 0 during [...]

Interesting idea about maybe not disengaging / re engaging the delay while processing delay automation when it hits 0 (eg keep the delay active)

Just out of curiosity if I find time I’ll look into it

My assumption without digging into the code at all:

I have a feeling you’re right that when delay hits 0, it discards the delay buffer and disables the delay 

Then when it goes above 0 again it reallocates the delay buffer and re engages the delay

Considering the grain buffer causes a click on activation, this is probably a “minor” version of that, prob much smaller buffer to allocate, hence much smaller click

TheHypnotist333, 2025-05-17:
Thank you
I'm glad to hear you intend to improve the sound engine in the long term.

ok-reza's fix is clever. I was able to remove the clicking by going into the automation view after recording and pulling up all spots where the amount is at zero. This is definitely a fix, it's not ideal in many cases but at least I know what the actual problem is and can work around it. Thank you!

One of the first effects I always reached for is the delay amount knob so you can imagine how I go to start a song and within 30 seconds im running into clicking problems wondering what in the world is going on. I cant be the only one who has tried to fiddle with the delay knob during recording. If this cant be solved I believe it's worth mentioning in the manual or somewhere public, I wish I had this information before. I swear I ran into this in almost every song all these years just because I'm a delay type of guy!

seangoodvibes, 2025-05-17:
> Thank you
> I'm glad to hear you intend to improve the sound engine in the long term.
>
> ok-reza's fix is clever. I was able to remove the clicking by going into the automation view after recording and [...]

Interesting 

Thanks for testing 

I'll try and follow the clues here and see why dropping to "0" has this impact on the sound

We might be able to devise a solution around that

TheHypnotist333, 2025-05-17:
Wait, I tried it in another project and the issue is still there. I'm not sure why it worked in the other one. It seems to do with the speed of modulation, is that what you were talking about with the buffer? I imagine me pulling up the zeroes slowed the movement of automation enough to stop the clicks in the other one I tried but not this one.

seangoodvibes, 2025-05-17:
> Wait, I tried it in another project and the issue is still there. I'm not sure why it worked in the other one. It seems to do with the speed of modulation, is that what you were talking about with [...]

Ah dang, yea I think that’s what I meant by the buffer

The size of the memory buffer required for the delay gets adjusted extremely frequently which I think causes clicks when it needs to reallocate a big enough amount that it interrupts the audio render.

The thing about the deluge is that it is not a multi threaded device. It has one thread that runs everything. So if one things takes a long time to complete, it has the potential of throwing other things off, such as the audio rendering.

We’ve gone to great lengths to fine tune things, and one of our dev’s built a genius scheduling system to help tremendously, but it’s very difficult to get the same performance with a single threaded device compared to a multi threaded device.

TheHypnotist333, 2025-05-18:
Do you have any ideas why I might he encountering this much more than other users aside from me loving to automate delay? I notice how you and the other dev said you cant reproduce this from a default song but I can. I wish some developer could watch me try to make a beat and see how often this happens for me, I really wonder if its the same for everyone or if my machine is especially prone to this stuff. I promise this is a constant issue for me and the fact hardly anyone else is naturally encountering this is so mystifying to me.