SynthstromAudible/DelugeFirmware #3746: Crash when looping and undoing loops on midi tracks.

MrS3th, 2025-05-22:
### Please describe the problem:

The Deluge crashes after changing presets on my studiologic midi keyboard (SL88) and then looping MIDI in Grid view, as described in the provided steps below. The crash occurs reliably when I undo the MIDI clip I've just looped, but the Deluge also crashes at some point or just freezes when I keep the MIDI clip playing or if I copy the MIDI clip elsewhere.

The changing of presets on the SL88, as far as I can tell, should not interact at all with the Deluge, but it apparently does. The presets split the keyboard in up to 4 zones, sending them to potentially different destination, such as different MIDI channels over USB. I could not find anywhere infos about what messages the SL potentially sends when turning the central knob. Regardless of the message though, it should not lead to a crash :) .

If that helps at all, another erratic behavior I observed, with the change of preset on the SL88 is the following (which might be a related issue or might require its own bug report). When the Differentiate midi device option is unchecked and I midi learn master volume for synth clips on a fader on another MIDI controller on channel 1 (nanoKONTROL2), and the SL88 is also midi learned for playing notes on the same clip, turning the preset knob on the SL88 resets the clip's volume to 0, regardless of previous level, be it set by the controller or the Deluge's knob. I've "solved" the behavior by turning off MSB (most significant bit) and LSB (least significant bit) on all zones for all presets on the SL88. Changing preset does not affect volume anymore, not that it should ever had.

### What is the expected behavior?

The Deluge should not crash when interacting with a MIDI track.

### When did you start noticing the issue?

A week ago, when I installed BETA.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-1BEB424

### If possible provide the steps to reproduce the issue and upload additional media:

1. Connect the SL88 to Deluge.
2. Turn the Deluge on.
3. New song.
4. Create midi clip in grid view.
5. Midi learn midi clip to SL88.
6. Turn the knob to change presets on SL88.
7. Play.
8. LOOP (global command).
9. Play some notes.
10. LOOP.
11. Undo.
12. Freeze or crash.

[image]

m-m-adams, 2025-06-19:
> If that helps at all, another erratic behavior I observed, with the change of preset on the SL88 is the following (which might be a related issue or might require its own bug report). When the [...]

MSB and LSB are CC 0 and 32. If you learned one of those CCs to volume this would be the expected behaviour

m-m-adams, 2025-06-19:
are you looping audio or midi?

seangoodvibes, 2026-04-19:
Hi @MrS3th can you please confirm if you can reproduce this issue with the latest nightly build? Thanks!