SynthstromAudible/DelugeFirmware #3768: Looping sample playback does not work in 1.3.0 Beta

jceloi, 2025-05-24:
### Please describe the problem:

When using the sample engine for a synth and setting loop points in the waveform menu the playback goes from start to end without looping.

### What is the expected behavior?

The sound should loop between the loop points.

### When did you start noticing the issue?

Tested on 1.3.0 BETA-AC0622B, does not work as expected. Tested back on 1.2.1 : works as expected.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0 BETA-AC0622B

### If possible provide the steps to reproduce the issue and upload additional media:

Video visibles here : https://discord.com/channels/608916579421257728/1138924035657449482/1375918932568965160

pmatila, 2025-10-03:
@stellar-aria @m-m-adams : I bisected this down to PR #2944 (95311abaf7969ee4ad2cd5b78d69e47e97f3723c), and further down to this commit within the PR:

commit 5c5d9569710b888fad58118ce2b874d9b8b9aa49 (HEAD)
Author: Katherine Whitlock <kate@skylinesynths.nyc>
Date:   Wed Nov 20 00:49:55 2024 -0500

    Sound as reference

pmatila, 2025-10-04:
Spent a good while staring at that patch, couldn't see anything obviously wrong, but in my (limited) experience with references is that if there's a bug, it's a missing `&` in some *non*-obvious place. So I started redoing the patch in tiny pieces to pinpoint it.

This is a total WTF, but it's 100% reproducible for me over multiple builds and reboots and all: it's the `const` added to getLoopStartPlaybackAtByte(). Dropping that fixes the sample looping (at least my testcase!) in community, 1.3 and immediately after the PR was merged.

With the const, getLoopStartPlaybackAtByte() always returns the value 44 for all looping samples I threw at it, so it basically appears to loop the entire sample instead of the defined loop point. With the const removed, the value changes per sample as you'd expect.

It doesn't make any sense whatsoever, so smells like UB I guess? I can certainly submit a PR, unless somebody wants to dig deeper at the issue.

pmatila, 2025-10-04:
Oh... there's no mystery afterall. The "const" makes the function signature differ from the version in SamplePlaybackGuide, but since there's no "override" specified in the VoiceSamplePlaybackGuide version, the compiler doesn't complain but goes to pick up the SamplePlaybackGuide version. I'm sure this is a perfectly logical choice, if you're a c++ compiler 😆 

I'll submit a PR.

PR #4109 (2025-10-04): Fix looping sample playback regression (#3768)
One of the changes in commit 95311abaf7969ee4ad2cd5b78d69e47e97f3723c was making getLoopStartPlaybackAtByte() and getLoopEndPlaybackAtByte() of VoiceSamplePlaybackGuide const, but this causes their signature to differ from the parent class SamplePlaybackGuide versions. Which in the case of getLoopStartPlaybackAtByte() at least, causes the wrong (parent class) version to be called, and causing the samples to loop from the beginning instead of their designated loop starting point.

Make the SamplePlaybackGuide versions const as well to get the signa [... cut]
commit ddb9e054 (2025-10-04): Fix looping sample playback regression (#3768) (#4109)

One of the changes in commit 95311abaf7969ee4ad2cd5b78d69e47e97f3723c
was making getLoopStartPlaybackAtByte() and getLoopEndPlaybackAtByte()
of VoiceSamplePlaybackGuide const, but this causes their signature
to differ from the parent class SamplePlaybackGuide versions. Which
in the case of getLoopStartPlaybackAtByte() at least, causes the
wrong (parent class) version to be called, and causing the samples to
loop from the beginning instead of their designated loop starting point.

Make the SamplePlaybackGuide versions const as well to get  [... cut]
closed 2025-10-04 by seangoodvibes