SynthstromAudible/DelugeFirmware #3809: Crash with E078 while browsing presets

stosem, 2025-06-03:
### Please describe the problem:

Deluge freezes with error `E078` when I browse Boards of Deluge 7 presets (which are heavy FX and are mostly sample based). I basically play a few notes and select the next preset, then Deluge crashes on a random preset.
Confirmed in the latest beta 1.3.0-880AE4A, but also exists in alpha and previous versions.

[image]

### What is the expected behavior?

Deluge shouldn't crash.

### When did you start noticing the issue?

just now and before

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-BETA-880AE4A

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turned on
2. Created a new synth clip
3. Load + Synth
4. Go to BoD7
5. Choose a preset
6. Play a few notes
7. Switch to next
8. Repeat until Deluge crash with error (usually after 10-15 presets).

What other details could I provide that would be helpful?

m-m-adams, 2025-06-17:
If you can get it to happen reliably then sharing your steps might help find a fix but otherwise just use 1.2.1 and it won't crash. There's nothing user facing to debug and we don't know the root cause

m-m-adams, 2025-06-19:
As far as I can tell this is fixed in nightly by #3617 

We aren't going to be able to fix it in beta as this has been around for years and needed a large refactoring. The freeze on this error code will be disabled automatically when the beta is changed to a release candidate

Metamere, 2025-07-26:
This issue seems to cause a crashes consistently and much quicker when browsing presets while creating a synth row in a kit, i.e. in the SYNTH TO ROW menu. waiting for several multisample presets to load seems to cause the crash more quickly.