SynthstromAudible/DelugeFirmware #3822: Chrod mode doesn't always work properly

sdesign75, 2025-06-05:
### Please describe the problem:

The chord mode isn’t always working on newly created clips. Seems intermittent. Sometimes it’s works sometimes it doesn’t. When you select a chord type and play the keyboard it’s not applied. Also, sometimes when a chord type is selected and I can play the chord on the keyboard, if I deactivate the chord then the selected chord type still plays. So it’s not being deactivated.
To summarise, sometimes you can select a chord type and it isn’t applied, and sometimes you can deactivate a chord type but it still plays
Changing the chord type does change the chord type either. It’s locked on the selected chord

### What is the expected behavior?

See above

### When did you start noticing the issue?

beta 1.3

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

m-m-adams, 2025-06-19:
I've tried to reproduce this but can't using any of the chord sidebar, chord library, and chord keyboard. Could you provide more detailed steps?

sdesign75, 2025-06-20:
Does this help?

https://discord.com/channels/608916579421257728/1118654563596128336/1378417679479603273

m-m-adams, 2025-06-21:
It  doesn't, if I do the same thing the chord doesn't get stuck