SynthstromAudible/DelugeFirmware #3840: MIDI automation overload over DIN and USB MIDI OUT

jasonhischier, 2025-06-13:
### Please describe the problem:

I occasionally update to the latest nightly firmware and am not concerned if I have to rollback if needed; in early April, I noticed that songs with a handful of MIDI automation lanes cause issues during playback. I rolled back my firmware to the most recent nightly version I could find (March 4), and I no longer had any issues. Fast forward to a couple days ago: I installed the June 11 nightly firmware and experienced the playback issues again. After scouring the Deluge discord and code/PRs/issues in this repo, I'm curious if #3531 is perhaps causing the issue.

Some notes about my setup / the use case:

- my Deluge received the OLED upgrade a year or so ago
- I can reproduce this issue whether Interpolation is on or off for the MIDI clip
- I can reproduce this issue whether the automation was recorded live via gold Deluge knobs
- I can reproduce this issue whether the automation was recorded live via sending MIDI CCs via DIN MIDI
- I can reproduce this issue whether the automation was recorded live via sending MIDI CCs via USB MIDI
- I can reproduce this issue with Deluge sending MIDI to a hardware synth over USB Host
- I can reproduce this issue with Deluge sending MIDI to a hardware synth over DIN MIDI connected via Deluge's DIN MIDI OUT
- I have tested this with DIN MIDI to EssenceFM, ASM Hydrasynth, and Beetlecrab Vector Synth, plus a homemade Teensy 4.1 synth over USB MIDI
- actions using the Load button to bring up the song load menu or load a synth/kit/etc while playback is occurring can slow the Deluge to a halt

Please let me know if there are any workarounds I may be able to try. I'm too busy w/ life lately to dive into the code myself to try to debug/make changes (I'm a developer; I've previously cloned/built the repo to explore adding some features), otherwise I'd spent more time doing that.

Thank you! Love all the updates/features that have been pouring in!

### What is the expected behavior?

MIDI instrument playback does not lag behind when there are a handful of automation lanes sending MIDI CCs.

### When did you start noticing the issue?

- nightly firmware around 4/6/2025, but the latest firmware I have that works is from 3/4/2025.

### Is there a relevant Pull request?

3531? Maybe?

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.3.0-NIGHTLY-EEB9FB5

### If possible provide the steps to reproduce the issue and upload additional media:

I've attached a zip containing 3 files:
- Nightly firmware from 3/4/2025
- Nightly firmware from 6/11/2025
- a Deluge XML song to demonstrate the issue (Midi Autm Overload.XML)

[Midi Automation Overload Issue.zip](https://github.com/user-attachments/files/20733341/Midi.Automation.Overload.Issue.zip)

**Steps to reproduce**
1. Install the 3/4/2025 firmware
2. Load Midi Autm Overload song (it has a 1 bar clip for channel 7 and automation on the first few MIDI CCs)
3. Enable metronome playback on the Deluge
4. Play the song and you should hear a melody of 4 quarter notes playing on the beat
5. Install the 6/112025 firmware
6. Load Midi Autm Overload song
7. Play the song:

**Expected Result**: the song plays the melody of 4 quarter notes playing on the beat

**Actual Result**: song playback drifts behind the metronome

seangoodvibes, 2025-06-14:
@jasonhischier thanks for reporting

can you confirm whether this problem persists in the 1.3 beta build?

The nightly has a lot of bugs while the 1.3 beta has a lot less.

That said, if the interpolation feature is the culprit, then it should be present in 1.3 as well.

Thanks

m-m-adams, 2025-06-17:
This has prompted me to notice that midi interpolation only works on mod wheel/pitch bend/aftertouch. I'm opening a PR to fix that but it seems unlikely to be causing your problems