SynthstromAudible/DelugeFirmware #3883: Pressing a letter pad on the keyboard does not enter this letter into sample browser menu on the first time

lburygin, 2025-06-25:
### Please describe the problem:

If I open the sample browser menu and try to search something using the keyboard, the first time I press a letter on the keyboard it doesn't get entered into search line. I need to press it again and then it's entered.
This doesn't happen on 1.2 fw

### What is the expected behavior?

Pressed letter pad should enter the letter first time

### When did you start noticing the issue?

today

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

866edb8

### If possible provide the steps to reproduce the issue and upload additional media:

1. Open the sample browser menu
2. Press 'A' letter pad to search something starting with 'A' letter

Metamere, 2025-08-16:
It seems that this applies to any browser menu where the keyboard is not shown from the start, so the load song menu has this issue as well.

lburygin, 2025-12-05:
I don't think it's intentional since it works differently in 1.2 fw, why could it be changed?

closed 2026-01-06 by seangoodvibes