SynthstromAudible/DelugeFirmware #3898: Crash when exiting Settings (when tapping quickly Back several times)

soymonitus, 2025-07-03:
### Please describe the problem:

If you press Back quickly while you are exiting Settings menu, it crashes

bash-3.2$ arm-none-eabi-addr2line -Capife build/Release/deluge.elf 0x2013507d 0x20114be5 0x20107f7f 0x201085f5
0x2013507d: disk_read.isra.0 at ??:?
0x20114be5: Userdef_INTC_HandlerExe at /Users/admin/Development/Projects/Deluge/DelugeFirmware/src/RZA1/intc/intc_userdef.c:734
 (inlined by) INTC_Handler_Interrupt at /Users/admin/Development/Projects/Deluge/DelugeFirmware/src/RZA1/intc/intc_handler.c:135
0x20107f7f: move_window at /Users/admin/Development/Projects/Deluge/DelugeFirmware/src/fatfs/ff.c:1070
 (inlined by) move_window at /Users/admin/Development/Projects/Deluge/DelugeFirmware/src/fatfs/ff.c:1056
0x201085f5: dir_find at /Users/admin/Development/Projects/Deluge/DelugeFirmware/src/fatfs/ff.c:2433

### What is the expected behavior?

No crash

### When did you start noticing the issue?

Latest beta

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

ok-reza, 2025-07-03:
possibly related: #2759

seangoodvibes, 2025-11-22:
Same issue reported on discord today:

[image]

kastenbalg, 2026-03-14:
Crash I reproduce is similar location in source as original report.

arm-none-eabi-addr2line -Capife build/Release/deluge.elf 0x20140599 0x20146a44 0x2011530f 0x201159b5
0x20140599: disk_read.isra.0 at ??:?
0x20146a44: icciar_read_complete at C:/Users/wilbo/source/repos/DelugeFirmware/src/RZA1/compiler/asm/irqfiq_handler.S:161
0x2011530f: move_window at C:/Users/wilbo/source/repos/DelugeFirmware/src/fatfs/ff.c:1070
 (inlined by) move_window at C:/Users/wilbo/source/repos/DelugeFirmware/src/fatfs/ff.c:1056
0x201159b5: dir_find at C:/Users/wilbo/source/repos/DelugeFirmware/src/fatfs/ff.c:2433

kastenbalg, 2026-03-14:
Think we tracked it down and fixed it.

https://github.com/CaptPat/DelugeFirmware/issues/55
https://github.com/CaptPat/DelugeFirmware/pull/53/