SynthstromAudible/DelugeFirmware #3943: Crash changing kit row to midi

macbutch, 2025-07-10:
A few notes. This seemed fairly easy to reproduce:

1. Add a kit row below the other rows
2. Change it to midi without loading a sample first

When it crashed, I was able to save and then reload. It had the new kit row (with no sample) and then crashed again when I tried the same operation. 

Creating a kit row at the top seems to work fine. 

[image]

Metamere, 2025-07-24:
I can't seem to reproduce this issue. Is there a kit you can send that this happens on?

Metamere, 2025-07-26:
Alright, with my PR #3977 merged, this issue should be fixed and can be closed.

nikodemus, 2025-08-07:
@macbutch can you confirm this is no longer reproducible, or provide exact step-by-step reproduction?