SynthstromAudible/DelugeFirmware #4020: Deluge crashes/freeze when using performancemode effects while recording in arranger

Anemoenchen, 2025-08-12:
### Please describe the problem:

when i use Performance effects while recording in arranger the deluge crashes. 

### What is the expected behavior?

no crashing

### When did you start noticing the issue?

"just now

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Beta, Built from community

### If possible provide the steps to reproduce the issue and upload additional media:

record into arranger over the shortcut "record + song" and then use performance effects for example the bitcrusher
after that the deluge crashes

solaris76, 2025-08-25:
Same issue here when recording CC MIDI automation.

jatomlin, 2025-10-02:
Was able to reproduce on Deluge nightly Release 20250930, but so far only via adjustment of the delay amount during record into arranger. Also got an error "E445".