SynthstromAudible/DelugeFirmware #4080: Activating Chord mode in 1.3 makes scale mode unstable

Dubprince, 2025-09-10:
### Please describe the problem:

When in scale mode, activating chord mode turns off scale mode. I realised from watching videos about the feature that the scale light has always been turned off in chord mode but not in chord library. Also in chopin. The problem is that it doesn't turn back on in chord library mode in 1.3. Not even after exiting keyboard mode can I turn it back on.
The weird thing is if I scroll trough the keyboard modes counter-clockwise the scale mode in chord library turns back on and I can't turn it off anymore. If I scroll clockwise the scale mode in chord library turns off and I can't turn it back on. In both cases it doesn't matter in which mode, the scale button is not functional anymore.

### What is the expected behavior?

The scale button stays functional after entering chord mode

### When did you start noticing the issue?

After using the 1.3 firmware.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.3.0-Beta-bea9a7c

### If possible provide the steps to reproduce the issue and upload additional media:

1.Start a new song and the scale mode will be on by default.
2.Press the keyboard button
3.Hold the keyboard button and start turning the select knob clockwise until you reach chord library. The scale button light will turn off as soon as you reach chord mode and you won't be able to turn it back on when you reach chord library mode.
4. Press keyboard button to exit keyboard mode.
5. Press the scale button and it won't work.
6. Press the keyboard button again.
7. Hold the keyboard button and this time scroll counterclockwise. You will see that the scale button in chord library mode will be lit.
8. Exit keyboard mode. Now you won't be able to turn off scale mode.

MrHaila, 2025-12-14:
Another repro of unexpected behaviour:

1. New song with empty clip
2. Hold keyboard and use select to go right by three click to activate chord keyboard. Observe that the scale button light turned off automatically.
3. Exit the keyboard mode by pressing the keyboard button. Observe that the piano roll is now in chromatic mode, scale button light is off.

Variations of this same thing is holding the keyboard button and navigating between the keyboard modes. The internal transposition mode gets inherited from the previous keyboard screen instead of reverting to what the user had selected originally.

seangoodvibes, 2026-10-16:
"The problem is that it doesn't turn back on in chord library mode in 1.3. Not even after exiting keyboard mode can I turn it back on."

This is fixed now. If I scroll layouts to chord mode scale turns off. then if I scroll right again to chord library, scale remains off but I can now toggle it on and off again by pressing the scale button.

regarding chord mode turning off scale, this seems to be intentional. Chord mode builds the layout from the song's scale. 

It turns off scale so that when you record notes from the layout it won't accidentally change that scale which can happen with column mode.

While recording:
- Row mode builds notes from the selected scale, so ordinary in-scale playing should be fine.
- Column mode uses preset chord voicings, some of which can contain out-of-scale notes.

So since recording with column mode could record out of scale notes which would change the songs scale to a custom one, the creator of the feature most likely disabled it as a precaution.

closed 2026-10-16 by seangoodvibes